mavlink/qgroundcontrol #6869: testHighDPI problem in ScreenToolsController

Sidewinder1138, 2018-09-13:
I'm looking at ScreenToolsController.h (latest sources from github), and I noticed something odd starting on line 91:

```
#ifdef QT_DEBUG
    bool testHighDPI            () { return qgcApp()->testHighDPI(); }
    bool isDebug                () { return true; }
#else
    bool isDebug                () { return false; }
    bool testHighDPI            () { return false; }
#endif
```

It seems to me that the testHighDPI function shouldn't return false always when in release mode, right? I can't imagine why that would be desired...

Also, it appears that testHighDPI isn't really used anywhere in the code? At least, I wasn't able to find any actual usage of it. I'd like to know how QGC handles highDPI in general...

dogmaphobic, 2018-09-13:
No, it isn't used anywhere. @DonLakeFlyer added that a long time ago just for testing and dealing with high DPI displays. All that code could probably be safely removed. With that said, it's best to wait until he's back in a couple of weeks.

DonLakeFlyer, 2018-09-25:
The code is correct although no longer used. This was a command line option used in debug builds only which would bump up the font size to simulate a high dpi screen (if I'm remembering correctly).

PR #6883 (2018-09-25): Remove unused code
Fixes #6869
closed 2018-09-25 by DonLakeFlyer