mavlink/qgroundcontrol #6872: idea for improved geotagger

Antiheavy, 2018-09-16:
We sometimes use the geotagger in QGC when we fly consumer cameras (eg Sony). I think there is room for improvement. Specifically a geotag mode that tries to match the image file times to the logfile lat/Lon/alt.

My thought is a button/option in the QGC geotagger window: “sync image times to first image trigger”

This option will sync the image file time (created by the camera) of the first image to the log file time of the first trigger (GPS time? or time since boot?  doesn't matter in this case). Then all subsequent image save times can be synced to the exact metadata in the log file.

This would be based on time since the first trigger, ignoring the rest of the image trigger commands in the log.  Another benefit is this solves the problem of missed triggers or deleted photos from the camera.

An option for an estimated image capture delay time could be added too.

@birchera @tubeme @LorenzMeier tagging you guys because of recent geotag/event log discussions elsewhere.  e.g. 
https://github.com/PX4/Firmware/issues/10189
https://github.com/mavlink/qgroundcontrol/issues/6772
https://github.com/PX4/Firmware/issues/10188

tubeme, 2018-09-22:
@Antiheavy I think we don't need this feature. Because we need the time stamp up to the moment of parsing. After that the files contain their POS data so timing is irrelevant. As user I don't care about timing and syncing etc. I just need my images with POS data and the CSV file for the Pix4D.

Antiheavy, 2018-10-24:
The following recent improvements in PX4 are probably better than my idea above.  Just a matter of someone implementing them in QGC.  ;-)

@beat  nice improvement: https://github.com/PX4/Firmware/pull/10738

@sanderux  nice improvement: https://github.com/PX4/Firmware/pull/10716

moreba1, 2020-03-05:
@Antiheavy 
@DonLakeFlyer 
Can you save all positions in one text file? 
image name - lat - long - alt

closed 2026-10-06 by github-actions[bot]