mavlink/qgroundcontrol #6875: Source Code: Text Styling and Formatting

tops4u, 2018-09-17:
While working on Translations on Crowdin I stumbled over the following Text:

`Clicking “Apply” will save the changes you have made to your airframe configuration.&lt;br&gt;&lt;br&gt;All vehicle parameters other than Radio Calibration will be reset.&lt;br&gt;&lt;br&gt;Your vehicle will also be restarted in order to complete the process.`

IMHO HTML Tags should not be used in Texts. Some Texts are broken using conventional <CR>, others are split over multiple Text Elements, and the example above is using HTML Tags. 

Are there Guidelines on how to Style/Format Text?

Above example is from: /src/AutoPilotPlugins/PX4/AirframeComponent.qml:107

DonLakeFlyer, 2018-09-25:
Styled text is used in a number of places in the UI. Not a lot, but certainly in mulitple places. I get how this complicates translation. But can't be changed with code changes.

> Some Texts are broken using conventional , others are split over multiple Text Elements, and the example above is using HTML Tags.

Not quite sure what you mean here?

DonLakeFlyer, 2018-09-25:
@dogmaphobic Any ideas here. The inline formatting stuff can be pretty valuable in places. Not using it and keeping the formatting featured requires all sorts of gymastics on the code. And also forces the positioning of where the highlights would be with respected to split up paragraphs/sentences of text.

tops4u, 2018-09-26:
@DonLakeFlyer Regarding your Question. I have seen Texts been broken into multiple Lines or Blocks of Text by using HTML Tags, Split the Text into multiple Elements or by adding CR/NL in the Text itself. So there are three variants of achieving the same styling.

DonLakeFlyer, 2018-09-28:
I talked with Gus about this. The way QGC does this with the inline tags is a pretty common thing to do. It's basically the same issue that comes up when translating web page content. So the answer is to translate the string and keep the same tags around the words in the translated string that match the placement in the english string such that the same concept gets across to the user.

closed 2018-09-28 by DonLakeFlyer