mavlink/qgroundcontrol #6878: Scale is working erratically

Williangalvani, 2018-09-20:
When resizing the window vertically, the scale changes even though the map does not:
[image]

Tested on master in Linux and 3.4.0 in Windows

dogmaphobic, 2018-09-20:
Interesting, this has probably been there since the beginning and no one ever caught it. I'll look into it.

DonLakeFlyer, 2018-09-25:
Back into Stable if simple enough fix.

dogmaphobic, 2018-09-25:
Well, it's all correct after all. The issue is that the code measures the distance between two points using where the scale is on the map. In the example above, you are moving the scale between Antartica and half way towards South Africa. The scale is showing the correct distance. It changes because of the map projection. This "error" is only seen if the change due to the projection is very large. On a "normal" use case, where your scale would be showing meters rather than thousands of kilometers, the difference due to (spherical) projection is so small to be insignificant.

closed 2018-09-25 by dogmaphobic

Williangalvani, 2018-09-25:
Of course! That makes perfect sense, my bad!