mavlink/qgroundcontrol #6882: Expose Mavlink packet loss data to the UI

DonLakeFlyer, 2018-09-25:
There currently is no way to see if your comm link is crap.

dogmaphobic, 2018-09-25:
I've added some to the _MAVLink_ panel:

[image]

This shows a particularly bad case when using WiFi. QGC is trying to load parameters while being bombarded with telemetry at the same time.

I've been talking to @dagar about this. There should be a way to tell the autopilot to pause/throttle back on telemetry when QGC first connects (and a whole lot of other things that can be done to help data traffic congestion).

DonLakeFlyer, 2018-09-25:
Ah, awesome. Is this on dailyt only?

dogmaphobic, 2018-09-25:
Correct.

DonLakeFlyer, 2018-09-25:
Nice, thanks