mavlink/qgroundcontrol #6888: How to get tlog from iPad?

MyOnlyOnlySunshine, 2018-09-26:
On iOS. device ,I can not get tlog，could anyone tell me the way to get it ? Thanks

hamishwillee, 2018-09-26:
If it isn't accessible from [Analyze > Log Download](https://docs.qgroundcontrol.com/en/analyze_view/log_download.html) then it probably is only possible to download logs from QGC on desktop builds. 

You can extract the memory card from the vehicle and read it on your computer.

MyOnlyOnlySunshine, 2018-09-26:
yeah，but it's tlog or ulog ？

hamishwillee, 2018-09-26:
It doesn't matter, since if the log download isn't supported it isn't supported.
If you're working on PX4 then a TLOG won't even be generated.

DonLakeFlyer, 2018-09-26:
> If it isn't accessible from [Analyze > Log Download](https://docs.qgroundcontrol.com/en/analyze_view/log_download.html) then it probably is only possible to download logs from QGC on desktop builds.

That is correct. None of the Analyze options are available on Mobile builds. Some of them may show up at some point but not sure when (there is an issue feature request in for that already). That said, even if they did, on iOS downloading over a WiFi connection would take forever.

closed 2018-09-26 by DonLakeFlyer