mavlink/qgroundcontrol #6890: Unable to retrieve full set of parameters with telem.

tuloski, 2018-09-26:
Using latest stable qgroundcontrol and pixracer with latest px4 stable firmware.

When connecting via telem1 on a "custom" radio link qgroundcontrol is unable to get all the parameters. It gets let's say 70% then it stops saying "qgroundcontrol was unable to retrieve the full set of parameters".
When connecting via ESP WIFI it is able to get all the parameters succesfully.
The custom radio link has a baud of 38400 and no flow control.

I'm using this custom extras.txt to change the baud:

```mavlink stop-all
mavlink start -d /dev/ttyS1 -b 38400 -r 2000 -m minimal
mavlink start -r 20000 -b 921600 -d /dev/ttyS0```

tuloski, 2018-09-26:
I also tried without the "-m minimal" and changing the -r param from 800 to 4000 without any success.

DonLakeFlyer, 2018-09-26:
Using a daily build check your packet loss from the Mavlink page under Settings. If packet loss if high then parameter download can fail. And it indicates bad hardware setup which would require firmware/hardware folks to help you out since QGC is not involved in that.

tuloski, 2018-09-26:
What do you mean by daily build? Where can I check packet loss?

BTW mission planner doesn't complain and retrieves all the parameters, so it seems related to QGC.

DonLakeFlyer, 2018-09-26:
> What do you mean by daily build?

There are two version of QGC builds available. Stable builds and daily builds. Daily builds include all the newest features currently under development. If you go to qgroundcontrol.com and hover over Downloads there is a daily build link.

> Where can I check packet loss?

"check your packet loss from the Mavlink page under Settings". Settings is "Q" icon. (https://github.com/mavlink/qgroundcontrol/issues/6882)

tuloski, 2018-09-28:
There are very bad instructions to build QGC with windows and I wasn't able.

But again, I don't think it's an harware issue since Misison Planner doesn't fail. Maybe there are different timeouts or whatever.

DonLakeFlyer, 2018-09-28:
> There are very bad instructions to build QGC with windows and I wasn't able.

Why are you trying to build QGC?

tuloski, 2018-10-01:
> Why are you trying to build QGC?

I found only the 3.4.4 as last release, which doesn't have that feature.
How can I use the master on windows without building it?

tuloski, 2018-10-01:
I found the daily build...and checked the loss rate.
It starts at 40% (when it loads well the parameters), then it get stuck and drops at loss rate of 95%.
I wonder how the sent messages are calculated on ground...

dogmaphobic, 2018-10-01:
Each message has a sequence number. The gaps are counted as dropped messages and QGC computes the number of messages sent from the autopilot based on this sequence number as well.

DonLakeFlyer, 2018-10-01:
At that loss rate things are just not going to work correctly. Seems like some sort of hardware problem.

tuloski, 2018-10-01:
I noticed that even if I started mavlink at 38400 baud and with -r 800 (800 B/s) I reach a frequency of PARAM_VALUE message of 60Hz, which is 1380B/s only the payload...with MAVlink overload we are close to 2000B/s only with that message. Probably that is killing (over buffering or whatever) my serial link.

What is actually limiting the PARAM_VALUE frequency? The MAVlink node onboard or it only answer to the GCS requesting the parameters (so the frequency is "controlled" by the CGS)?

DonLakeFlyer, 2018-10-01:
> What is actually limiting the PARAM_VALUE frequency?

Vehicle side. GCS does not request parameters individually initially, they are streamed by vehicle side down to ground station. Sounds like a firmware bug with its pipe filling rate determination code.

DonLakeFlyer, 2018-10-01:
@dagar Firmware bug?

dogmaphobic, 2018-10-01:
> I noticed that even if I started mavlink at 38400 baud and with -r 800 (800 B/s) I reach a frequency of PARAM_VALUE message of 60Hz, which is 1380B/s only the payload...with MAVlink overload and the other messages we are over 2000B/s. Probably that is killing (over buffering or whatever) my serial link.

@dagar Does parameter response honor the current set data rate? It seems it's streaming them out as fast as it goes. How about log download? It has the same issues.

dogmaphobic, 2018-10-01:
With that said, we need to be able to tell the autopilot to throttle down (pause) telemetry when QGC is initializing. The telemetry stream is only increasing bandwidth at that point.

tuloski, 2018-10-01:
> How about log download? It has the same issues.

I never tried because it's too slow to download logs, but I'll test it.

DonLakeFlyer, 2018-10-01:
> With that said, we need to be able to tell the autopilot to throttle down (pause) telemetry when QGC is initializing.

Why?

DonLakeFlyer, 2018-10-01:
> Does parameter response honor the current set data rate? It seems it's streaming them out as fast as it goes. How about log download? It has the same issues.

Firmware side has code to adjust rates to try to fill the pipe with data. Not sure if parameter streaming uses that to up the rate or not. Seems like it.

dogmaphobic, 2018-10-01:
> Why?

Because the parameter download is competing with telemetry. For a low bandwidth connection, if you are able to pause telemetry, you have the full bandwidth just for the initial parameter (and mission, etc.) download making it all a lot faster.

DonLakeFlyer, 2018-10-01:
Ah, I see makes sense. Have to be careful of heartbeats though. ArduPilot kind of does this, and heartbeats stop which means QGC has various special case code to turn off comm lost with ardupilot in these cases. If I remember correctly log download is one of those cases with ArduPilot. That ends up being a real pain and flaky.

dagar, 2018-10-01:
Issue created PX4 side. https://github.com/PX4/Firmware/issues/10618
The biggest immediate fix will be getting PX4 param send to respect the rates, but there are quite a few other areas that can be improved as well. Including things directly in QGC and the Mavlink spec.

@DonLakeFlyer @dogmaphobic happy to discuss ideas in detail if there's any time. In particular I think QGC asserting more control over the data streams would be quite helpful. By default even on a slow SiK radio we're sending about 100x more telemetry data than you need for typical usage.

shortcut21, 2018-10-31:
Hello, guys
I have troubles with the whole parameter list and also loading them in QGC takes a lot of time and eventually I get "GQC unable to retrieve full parameter list". How can I decrease the time of loading parameters into QGC and get the full parameter list up and running? I have pixhawk 3 pro and I run udp connection with telemetry radio and launch the script px4.launch. Any help is appreciated.

DonLakeFlyer, 2018-11-01:
If you are using a WiFi adapter then there was a recent fix made for a parameter loading bug.

dogmaphobic, 2018-11-07:
The code keeps track of "first messages" and uses the proper initial sequence number as an initializer:
https://github.com/mavlink/qgroundcontrol/blob/master/src/comm/MAVLinkProtocol.cc#L205-L211

tuloski, 2018-11-07:
@dogmaphobic that's weird, because as soon as I connect that counter is instantly at 1000+.

Also I tried with a "bridge" in between, i.e. a device that reads the telemetry and rewrites in a serial (everything cabled) and I'm calculating the received messages on the bridge and they are much less than the ones estimated there.

What do you do when the .seq reaches 255? I can't find it in the code.

closed 2026-10-06 by github-actions[bot]