mavlink/qgroundcontrol #6905: Configurable serial ports

dagar, 2018-10-01:
In PX4 (https://github.com/PX4/Firmware/pull/10337) we have different modules and drivers that can be configured to use serial ports. As a result it's currently possible to have a conflict configured via parameters.

On the QGC side I'd like to add a mechanism to configure the usage per port, where QGC handles the deconfliction.

### PX4
For example in PX4 we have the following parameters to configure the ports used by different mavlink instances or different GPS. Each parameter has the same drop down list of serial ports.

 * MAV_1_CONFIG
    * telem 1
    * telem 2
    * telem 3
    * gps 1
    * gps 2
 * MAV_2_CONFIG
    * telem 1
    * telem 2
    * telem 3
    * gps 1
    * gps 2
 * MAV_3_CONFIG
    * telem 1
    * telem 2
    * telem 3
    * gps 1
    * gps 2
 * GPS_1_CONFIG
    * telem 1
    * telem 2
    * telem 3
    * gps 1
    * gps 2
 * GPS_2_CONFIG
    * telem 1
    * telem 2
    * telem 3
    * gps 1
    * gps 2
 ...

### QGC
On the QGC side I'd like to invert the configuration and prevent conflicts. Each serial port should only appear in a single parameter.

 * telem 1
   * Mavlink 1
   * Mavlink 2
   * Mavlink 3
   * GPS 1
   * GPS 2
 * telem 2
   * Mavlink 1
   * Mavlink 2
   * Mavlink 3
   * GPS 1
   * GPS 2
 * telem 3
   * Mavlink 1
   * Mavlink 2
   * Mavlink 3
   * GPS 1
   * GPS 2
 * gps 1
   * Mavlink 1
   * Mavlink 2
   * Mavlink 3
   * GPS 1
   * GPS 2
 * gps 2
   * Mavlink 1
   * Mavlink 2
   * Mavlink 3
   * GPS 1
   * GPS 2

hamishwillee, 2018-10-05:
Just FMI, but would this be the only tuning page for those parameters, or would people still be able to see the underlying MAV_3_CONFIG, MAV_2_CONFIG etc? 
- If so, then what should the proposed inverted tuning page do if these are manually set with a conflict? 
- if not, how would the docs work, since these parameters are defined in our docs but would not be visible to users.

jafrado, 2019-02-18:
Would be great to have this ...

hamishwillee, 2019-02-18:
> As it stands, it's documented like this, but no way to configure com ports now:

@jafrado What do you mean - those documents show exactly how to configure them. It just isn't very intuitive.

All this issue asks is to be able to choose the associated service running on particular port (rather than starting selecting the port used by each service.)

jafrado, 2019-02-18:
Doesn't work with 1.8.2 and v3.5, tried today. As it stands I added mavlink commands to my extras.txt

hamishwillee, 2019-02-18:
@jafrado For 1.8.2 you would use SYS_COMPANION for this. In 1.9 and later you use the docs above. For a bit of guidance you can see the PR for the change here: https://github.com/PX4/px4_user_guide/pull/352/files

@dagar Can we run out versions for PX4 User Guide too so I can point to correct docs for this stuff?

jafrado, 2019-02-18:
Got it, thanks.