mavlink/qgroundcontrol #6910: Mavlink version mismatch error on MacOS

asimonov, 2018-10-03:
- Ground station operating system: MACOS 10.13.6
- QGroundControl version and build: 3.4.3
- Autopilot board: Pixhawk 2.1
- Autopilot firmware (e.g. PX4, ArduPilot, custom): PX4 1.8.0
- Exact steps to reproduce the problem: use TELEM2 to connect to NUC via USB, start mavros node on on-board NUC computer with gcs_url parameter pointing to MAC_IP (UDP protocol, default port). 

BUG: See the "Detected radio still using Mavlink v1.0..." (message from https://github.com/mavlink/qgroundcontrol/blob/master/src/comm/MAVLinkProtocol.cc#L298) in qgc in rapid succession.

The only way to stop these spurious pop-ups is this is to kill QGroundCountrol.

I see in the code that the message should not 'stick', implemented by increasing the counter. But it is not what happens.

DonLakeFlyer, 2018-10-06:
Can you turn on logging for MultiVehicleManager and attach the output here: https://docs.qgroundcontrol.com/en/SettingsView/console_logging.html

asimonov, 2018-10-10:
hi! sorry for late reply. did not have chance to re-run it on Mac. 

but seen this happen on Ubuntu version of qgroundcontrol v3.3.2 today. 
version 3.2.7 does not seem to have this problem. 

will keep the logging on and report when this happens again

asimonov, 2018-10-11:
[qgc-343-mavlink-error.txt](https://github.com/mavlink/qgroundcontrol/files/2469099/qgc-343-mavlink-error.txt)

[image]
[image]

asimonov, 2018-10-11:
the above if from MacOS version

asimonov, 2018-10-11:
just tried v3.2.7 and it definitely does not have this bug

asimonov, 2018-10-11:
just built latest master on MacOS and it definitely has the problem described above
attaching the screenshot of what the message looks like
[image]

DonLakeFlyer, 2018-11-08:
@asimonov Can you try building with this diff: 
[mavlink.diff.txt](https://github.com/mavlink/qgroundcontrol/files/2560146/mavlink.diff.txt)

I think the counter might be wrapping.

asimonov, 2018-11-10:
unfortunately I have moved jobs, so do not have access to reproduce

closed 2018-11-27 by DonLakeFlyer