mavlink/qgroundcontrol #6917: Airframe Setup (Disabled while the vehicle is flying)

blevien, 2018-10-09:
Just flashed px4 FMUv5 and QGroundControl v3.4.4 on MacOS

Cannot choose airframe, calibrate sensors, calibrate radio or flight modes, always getting the message below eventhough vehicle is disarmed and on the ground.

I installed QGroundControl a few versions old and was able to configure.

Still having issues with Auto Missions rejected, but will file separately if it is not related to a configuration issue.

[image]

DonLakeFlyer, 2018-10-24:
What version of Firmware and can you get me a screenshot of the EXTENDED_SYS_STATE mavlink message from the mavlink inspector.

Jetpackaviation, 2018-10-26:
It is happening to me as well. I am using px4fmu-v2_default for the hardware and QGC version 3.4 daily 10/26/18 daily build

DonLakeFlyer, 2018-10-29:
> can you get me a screenshot of the EXTENDED_SYS_STATE mavlink message from the mavlink inspector.

Need this.

DonLakeFlyer, 2018-11-08:
Can't do anything here without further information.

closed 2018-11-08 by DonLakeFlyer

masz106, 2018-11-09:
I have the same problem:
[image]