mavlink/qgroundcontrol #6918: Param loading issue with new mRo Pixracer MAVESP8266 WiFi

bkueng, 2018-10-10:
We got new Pixracers from mRo and there's a problem with param loading via WiFi. The WiFi seems to come with a new Firmware (http://firmware.ardupilot.org/Tools/MAVESP8266/latest/), easily noticeable with the changed WiFi SSID named ArduPilot.

The parameters load and show up on the parameter tab, but the Airframe shows as unconfigured, even though it's set.
This error message shows up, even at the beginning of loading params (the green progress bar for param loading starts at the same time the error message pops up):
[image]

I don't have this problem with an older MAVESP8266.
This is based off recent QGC master (around 6163ce95e3915edacb18bfc084d9285668266ba2).

Here are 2 QGC logs, the second one with the preflashed APM firmware (same issue):
[log.txt](https://github.com/mavlink/qgroundcontrol/files/2464902/log.txt)
[log_apm.txt](https://github.com/mavlink/qgroundcontrol/files/2464903/log_apm.txt)

dagar, 2018-10-15:
Any ideas @dogmaphobic?

dogmaphobic, 2018-10-17:
> Any ideas @dogmaphobic?

Not really, given that I've never heard of it. Where are the sources for this (ESP8266) firmware?

dagar, 2018-10-19:
@pkocmoud FYI

pkocmoud, 2018-10-19:
Thanks @dagar / Hi @dogmaphobic - I know Ardupilot has their own version of ESP01 firmware. I will test this out today.

DonLakeFlyer, 2018-10-24:
It seems like somehow param load was reported as complete while it was still trying to load parameters. Can someone send me one of these and I can debug.

pkocmoud, 2018-10-24:
They will be in the mail today. Thank you!

DonLakeFlyer, 2018-10-29:
There is a bug in QGC with respect to multi-component parameter fetch: If all the subsidiary components finish parameter download prior to QGC getting a single parameter from the main/default component then it will think download is finished. The new ESP8266 firmware moves around the ordering of parameter fetch such that this shows up for the first time with the ESP8266.

DonLakeFlyer, 2018-10-29:
While fixing this I discovered that the parameter editor is broken with respect to editing parameters from components other than the default component. I'll enter a separate issue on that. This won't be fixed on stable since it is too large a fix.

PR #6963 (2018-10-29): Fix multi-component parameter loading
Fixes #6918, #6941
closed 2018-11-07 by DonLakeFlyer