mavlink/qgroundcontrol #6924: Waypoint altitude wheel not useable on desktop

julianoes, 2018-10-16:
I'm trying to set up a mission with QGC 3.4.4 and setting the altitude suddenly has become really cumbersome. To lower the altitude, it takes an awful long time because I have to turn the wheel to the right. However, since my screen ends there, it takes many mouse movements.

Is there a trick I'm missing? Or a setting for non-touch screens?

[image]

julianoes, 2018-10-16:
Oh, I see. I can click on the thing to enter the number with the keyboard.
Could this option be made more prominent on desktop, or the wheel disabled for non-touch screen?

DonLakeFlyer, 2018-10-24:
> or the wheel disabled for non-touch screen?

I would think the wheel is kind of useful to be able to adjust by a few values without having to type.

> Could this option be made more prominent on desktop

Not quite sure what to do in order to do that. Editing with keyboard requires click which is the same as before. Any ideas?

julianoes, 2018-10-25:
> I would think the wheel is kind of useful to be able to adjust by a few values without having to type.

I disagree. I always need to change from e.g. 50 to 10 but never from let's say 50 to 48.

> Any ideas?

Disable it for desktop is my suggestion.

DonLakeFlyer, 2018-10-25:
> I disagree. I always need to change from e.g. 50 to 10 but never from let's say 50 to 48.

Ok fine. So don't use the wheel. Why disable? What problems does it create.

> Disable it for desktop is my suggestion.

How does that make it clearer that you need to click to edit the value?

DonLakeFlyer, 2018-10-25:
Closing. Without any ideas for how to make the need to click more obvious, or reason why mousewheel is bad it's good as is.

closed 2018-10-25 by DonLakeFlyer