mavlink/qgroundcontrol #6930: Command line options not working in Windows

Williangalvani, 2018-10-22:
The [Command Line Options](https://dev.qgroundcontrol.com/en/command_line_options.html) are not working in Windows.

The terminal returns with no output at all as soon as the process is launched, as if the process detached from it.

I tried 3.4.4 in two Windows 10 machines.

When building it from source, the debug build seems to work as expected, but the release build shows the same "detached" behavior.

DonLakeFlyer, 2018-10-24:
> The terminal returns with no output at all as soon as the process is launched, as if the process detached from it.

That's pretty much the way it works. Same for OSX if I remember. Only Linux will give you terminal output.

Williangalvani, 2018-10-25:
Ok, so is [DebugViewPP](https://github.com/CobaltFusion/DebugViewPP/releases) a good alternative to seeing debug info before a crash?
Or should people go straight to the .dmp route as explained [here](https://docs.qgroundcontrol.com/en/Support/Support.html)?

dogmaphobic, 2018-10-25:
> That's pretty much the way it works. Same for OSX if I remember. Only Linux will give you terminal output.

It works fine on macOS if you launch QGC from the command line. On Windows, I've always resorted to running it from within Qt Creator.

DonLakeFlyer, 2018-10-25:
What kind of debug info? For QGC log you can log to a file. For OS stuff you can use eventviewer to look at the crash or otherwise run using Qt Creator.

closed 2018-10-25 by DonLakeFlyer