mavlink/qgroundcontrol #6941: Parameter fetch fails with multiple MAVLink components

edwinhayes, 2018-10-25:
See also #6940 

QGroundControl fails to fetch the full set of parameters from the flight controller, if there are multiple MAVLink components talking (all of which respond to the broadcast parameter list request which QGroundControl sends).  If I filter out messages from components other that the flight controller, parameter fetching occurs normally.

Running daily build, looks like 06-09-2018, and latest beta Arducopter (although stable Arducopter v3.5.7 was the same, so probably unrelated).

dogmaphobic, 2018-10-25:
Are you using unique component ID for each of these components?

edwinhayes, 2018-10-25:
Hmm, I note that the message which says 'parameters are missing' appears way before the green progress bar has finished moving across the top of the screen.  Is that normal behaviour?  It seems unclear how it could reasonably conclude that it wasn't able to successfully fetch parameters before it's actually finished receiving them?

dogmaphobic, 2018-10-25:
> can you confirm that this definitely works in QGroundControl at the moment?

In all cases that I've tested with, yes. It doesn't mean that there could be something I have not seen before and missed something.

DonLakeFlyer, 2018-10-25:
Possibly related to #6918?

dogmaphobic, 2018-10-25:
> Possibly related to #6918 which also has the "pops error prior to param load finished" thing.

Could be but I don't know where the sources for that firmware is.

DonLakeFlyer, 2018-10-25:
Phillip sent me one. I'll debug it to figure out what is going on.

dogmaphobic, 2018-10-25:
All I need to is to look at what the code is doing. Is it "proprietary"?

DonLakeFlyer, 2018-10-25:
@tridge Where is the repo for you ESP8266 firmware?

DonLakeFlyer, 2018-10-25:
@dogmaphobic And just realizing you are probably not talking about this ESP8266 firmware!

dogmaphobic, 2018-10-25:
Correct. The question is valid but for the other issue (#6918). Not this issue :)

edwinhayes, 2018-10-25:
What is the condition or event which QGC determines that parameters have 'finished loading'?  Is it a timeout, or reception of the Nth parameters (where N matches the 'count' field)?  If I understood this, I could probably try some more combinations of MAVLink message filtering to see if I can narrow it down further?

DonLakeFlyer, 2018-10-25:
> Is it a timeout, or reception of the Nth parameters (where N matches the 'count' field)?

Both. QGC knows how many parameters it should get from each component. If it doesn't get all of them it goes into retry logic, querying individually multiple times. If that fails, it eventually gives up and reports params missing.

I would wait until I can debug #6918 and figure out what is going on there since that is a multi-component param case as well.

edwinhayes, 2018-10-26:
@DonLakeFlyer Ok thanks.

edwinhayes, 2018-10-28:
Most of my MAVLink components have far fewer parameters than the flight controller.  This means that QGC will probably see parameter 5/5 from one of the other components far sooner than it sees 850/850 from the flight controller.  Could this cause the 'Nth parameter' check to pass prematurely?

edwinhayes, 2018-11-04:
@DonLakeFlyer Tested and working.  Thanks.