mavlink/qgroundcontrol #6969: Add obstacles to survey grid

DangerD256, 2018-10-30:
I built pesticide sprayer, but current software don't allow me to add obstacles to grids. On fields we have trees or other obstacles, need to have ability to add obstacles as on dji software

Geryones, 2018-11-11:
Exactly the samething I need. Like an inverted fence. 
Also, if you could import a list of obstacles via a KML file!

mericakay, 2020-12-11:
I am working on the same issue, did you find a solution?

DangerD1024, 2020-12-11:
> I am working on the same issue, did you find a solution?

I just bought jiyi flight controller, it comes with app that has all crop sprayer functionaity

mericakay, 2020-12-11:
I understand I am developing qgroundcontrol. I will make this plugin. Thank you

DangerD1024, 2020-12-11:
> I understand I am developing qgroundcontrol. I will make this plugin. Thank you

Another problem is to make changes to arducopter firmware to add precise pump control (add water flow meter).

mohammada315, 2021-05-12:
@mericakay Hi, I'm also working on this
Did you reach a conclusion?

bousong, 2021-11-11:
Have you found the solution?

DangerD1024, 2021-11-15:
> I understand I am developing qgroundcontrol. I will make this plugin. Thank you

Maybe we can donate to someone together to make such feature?

closed 2026-10-06 by github-actions[bot]