mavlink/qgroundcontrol #6985: IOS - Ipad 10 inch No bluetooth

wx4cb, 2018-11-04:
Latest version of IOS, latest version of QGroundControl from the IOS store, when i try to connect to the crossfire module bluetooth (or even the BT on the aircraft itself)

When i try to setup the connection in the app, all i get is "Bluetooth not available".

I can connect perfectly fine to both modules in the settings app for the ipad, just cant "connect" via qgroundcontrol.

DonLakeFlyer, 2018-11-04:
@dogmaphobic Can you help with Bluetooth.

DonLakeFlyer, 2018-11-27:
@dogmaphobic Do you still have a crossfire?

wx4cb, 2018-11-27:
yes i do, I can help with both. What do you need me to do?

DonLakeFlyer, 2018-11-27:
Sorry, when you see a comment with @foo on the front of it that means the comment is targetted to that person.

wx4cb, 2018-11-27:
np, well the offer is still there :D

DonLakeFlyer, 2018-11-27:
Wait, is the Crossfire a registered licensed device for iOS? QGC only supports full bluetooth which requires the device to be iOS licensed.

dogmaphobic, 2018-11-27:
Yes, I have one. I haven't used it in ages. There is a work around if using BT BLE instead. However, last I checked Qt support for BLE was not working.

dogmaphobic, 2018-11-27:
I guess it needs some revisiting. I also don't quite get BLE. As far as I can tell, it wasn't meant for high bandwidth data streaming:

https://doc.qt.io/qt-5.11/qtbluetooth-index.html

wx4cb, 2018-11-27:
@DonLakeFlyer not sure if it's "ios licensed" - i'm not that well versed in the ios ecosystem, but it does work with mobileflight

DonLakeFlyer, 2018-11-27:
I don't think I'd fight with BLE. Regular Bluetooth is hard enough to support. Adding even more craziness doesn't sound productive.

cor-star, 2019-07-21:
Hi folks! Just want to chime in here that if you are at all open to looking into developing this issue.. it would truly be fantastic. An auxiliary ground station iPad would allow a visual observer to easily and seamlessly track our aircraft while we're flying, and QGroundControl is one of the more lovely apps available in the ecosystem. If you have ANY suggestion about what we might need to buy or do to be able to stream MAVLink to QGC, it would be of great interest..

DonLakeFlyer, 2019-08-13:
Android versions of QGC don't have this limitation with respect to bluetooth. iOS version of QGC is generally used in combination with a WiFi connection.

blazczak, 2020-10-18:
+1 requesting Bluetooth connectivity in the iOS version of QGC.

This has come up quite a bit before (including in #3583 before it was hijacked for Android, no offense). Also [here](https://discuss.px4.io/t/a-groundcontrol-bluetooth/6466/2) and [here](https://discuss.ardupilot.org/t/some-advice-for-qgc-ios-version/41737).

Can this be finally put to rest by either a working implementation or the QGC tech lead commenting in definite terms why this is not being pursued? The response `Apple requires that Bluetooth devices which connect to an Apple device are licensed by them. Which as far as I know nobody has done. QGC is not in control of this. This is not the case on Android.` is quite vague and perhaps even erroneous. Is the dev team saying that they’d have to submit hardware Bluetooth devices to Apple as part of the app review before you can put this feature up in the App Store? I don’t think this holds true, to follow the logic for instance music apps would have to ship to Apple every possible pair of headphones in existence to certify that their audio works over iOS Bluetooth. I seriously doubt they do. Maybe for custom interfaces and transports you’d have to do something like that, but what about simple serial over Bluetooth?

Tinkerers and enthusiasts would really dig having a serial-over-Bluetooth functionality present in QGC iOS, for example when using this simple hardware solution: https://www.dronetrest.com/t/bluetooth-to-telemetry-bridge/205 As an example, the HC-05 Bluetooth module connects and pairs with iOS just fine.

Another working example in iOS to show what is doable: http://kouky.org/blog/2016/04/09/mavlink-radio-telemetry-for-ios.html

wx4cb, 2020-10-18:
Bluetooth serial support is not going to happen in iOS until apple supports the SPP profile which is required for serial support.

DonLakeFlyer, 2020-10-18:
> Bluetooth serial support is not going to happen in iOS until apple supports the SPP profile which is required for serial support.

Correct. iOS only support BLE profiles. I thought there was some way to do it with licensed tech but maybe not.

blazczak, 2020-10-18:
Thanks for the quick replies. I didn't look deeper in Koukoullis' implementation (last link above) but in light of your responses how come then he was able to make it work over serial?

wx4cb, 2020-10-18:
Looking at his web page, he basically has something that converts it from serial to a protocol over BLE. I would hazard a guess it's not strictly serial.

Not having a BLE mini i cannot answer exactly

> Thanks for the quick replies. I didn't look deeper in Koukoullis' implementation (last link above) but in light of your responses how come then he was able to make it work over serial?

DonLakeFlyer, 2020-10-19:
I"m going to close this since it continues be the true that there doesn't seem to be any way to make this work.

closed 2020-10-19 by DonLakeFlyer