mavlink/qgroundcontrol #7003: NO RC MODO

jiaozhuliu, 2018-11-14:
![image](https://user-images.githubusercontent.com/36
[image]
[image]
  NO RC
288561/48465755-da10d800-e81e-11e8-8a32-8fe90c7f2671.png)

### ****

mrpollo, 2018-11-19:
Hey @jiaozhuliu looks like you are having trouble working with QGC, we are going to need more help in order to understand your problem, can you please fill in the following details:

## Expected Behavior
Please tell us what should happen as detailed as possible.

## Current Behavior
Please describe what happens instead of the expected behavior.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1.
2.
3.
4.

## System Information
When posting bug reports, include the following information
- Operating System: [e.g. iOS 12.1, Windows 8, macOS Mojave 10.14.1, Ubuntu 14]
- QGC Version: [e.g. 3.4.4]
- QGC build: [e.g., daily, stable, self-built from source, etc...]
- Flight Controller: [e.g., Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.]
- Autopilot (with version): [e.g., PX4 1.8.1, Ardupilot Copter 3.6.1]

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

DonLakeFlyer, 2018-11-27:
@jaxxzer ...

jaxxzer, 2018-11-27:
I see an apj file, we don't have any stable releases in apj form. @jiaozhuliu  Try the stable firmware: http://firmware.ardupilot.org/Sub/stable/

closed 2026-10-06 by github-actions[bot]