mavlink/qgroundcontrol #7005: Survey - Split Polygon: Hangs in split code

DonLakeFlyer, 2018-11-14:
* Load the attached shape into a Survey
* Switch to split polygons on
* Code will hang in the polygon splitting aprt

[Archive.zip](https://github.com/mavlink/qgroundcontrol/files/2582162/Archive.zip)

thomasgubler, 2018-11-24:
I can reproduce this. From some very brief debugging it seems never to reach the the termination condition of the recursion that splits the polygons.

feemi, 2018-11-26:
I also reproduced this when i load kml into survey
[sample.kml.zip](https://github.com/mavlink/qgroundcontrol/files/2616348/sample.kml.zip)

thomasgubler, 2018-12-14:
I plan to have a look over the holidays.

Next step is investigate if it is a special shape of polygons that triggers the issue or if it is just the number of vertices that make that part of the code run that long.

DonLakeFlyer, 2019-01-19:
I'm wrapping up Stable 3.5 so I'm going to need to turn this off in the code for now. Too late to get into 3.5.

thomasgubler, 2019-02-19:
[sample_min.kml.zip](https://github.com/mavlink/qgroundcontrol/files/2879558/sample_min.kml.zip)

Based on @feemi s file I was able to get a minimal kml where this breaks. In the zip there are the following files:
`sample_min_good.kml` : all is good; 13 vertices
`sample_min_critical.kml` : things start to get slow; 14 vertices
`sample_min_broken.kml` : it takes foerver; 15 vertices

Based on this I think this is an alogorithm performance problem (depending on shape and size of polygon) rather than a problem with the terminating condition.

DonLakeFlyer, 2019-02-19:
> Based on this I think this is an alogorithm performance problem (depending on shape and size of polygon) rather than a problem with the terminating condition.

Ick. That's kind of worse than a bug then.

closed 2026-10-06 by github-actions[bot]