mavlink/qgroundcontrol #7018: QGC v3.4.3 AutoLoad will not transfer geofence and rally points

hongminy, 2018-11-18:
This system is for posting bugs or feature requests ONLY. For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

----
When posting bug reports, include the following informaiton:
- Ground station operating system.
- QGroundControl version and build (daily, stable, self-built from source, etc.)
- **If you are using a Stable build which is not the latest, first step is to install latest Stable build and try again prior to entering a bug report.**
- Autopilot board: Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.
- Autopilot firmware (e.g. PX4, ArduPilot, custom) and version.
- Exact steps to reproduce the problem. Starting from booting QGroundControl to when the problem occurs.

**All this template text should be deleted before creating your issue.**

hongminy, 2018-11-18:
the plane will only load waypoints but not geofence and rally points

hamishwillee, 2018-11-18:
@hongminy Please fill in all that template text that you ignored above: https://github.com/mavlink/qgroundcontrol/issues/7018#issue-381913851. 

No one can debug an issue without enough information.

hongminy, 2018-11-19:
When posting bug reports, include the following informaiton:

QGC running on OS: OS X and Ubuntu and Windows
QGroundControl version: v3.4.4
Autopilot board: Pixhawk 2
Autopilot firmware PX4 1.8.1 stable
Exact steps to reproduce the problem: 1. Turn on the Auto Upload option in QGC, and put the file in the right directory. 2. launch QGC and then connect the Pixhawk. 3. The mission will be uploaded when the Pixhawk connects, but there are only waypoints in the mission plan view (no geofence and rally points).
PS: If I choose to manually upload the plan file, it does not have this problem even with the exact same plan file.

AutoLoad1.plan:
{
    "fileType": "Plan",
    "geoFence": {
        "circles": [
        ],
        "polygons": [
            {
                "inclusion": true,
                "polygon": [
                    [
                        38.14254379272461,
                        -76.43408966064453
                    ],
                    [
                        38.1418342590332,
                        -76.42526245117188
                    ],
                    [
                        38.144676208496094,
                        -76.42799377441406
                    ]
                ],
                "version": 1
            }
        ],
        "version": 2
    },
    "groundStation": "QGroundControl",
    "mission": {
        "curiseSpeed": 15,
        "firmwareType": 12,
        "hoverSpeed": 5,
        "items": [
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 10,
                "AltitudeMode": 0,
                "autoContinue": true,
                "command": 22,
                "doJumpId": 1,
                "frame": 0,
                "params": [
                    35,
                    0,
                    0,
                    0,
                    38.147918701171875,
                    -76.42799377441406,
                    10
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 60.959999084472656,
                "AltitudeMode": 0,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 2,
                "frame": 0,
                "params": [
                    0,
                    15,
                    0,
                    0,
                    38.14254379272461,
                    -76.43408966064453,
                    60.959999084472656
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 45,
                "AltitudeMode": 0,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 3,
                "frame": 0,
                "params": [
                    0,
                    15,
                    0,
                    0,
                    33.77146530151367,
                    -117.69239807128906,
                    45
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 33,
                "AltitudeMode": 0,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 4,
                "frame": 0,
                "params": [
                    0,
                    15,
                    0,
                    0,
                    33.770931243896484,
                    -117.69322204589844,
                    33
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 20,
                "AltitudeMode": 0,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 5,
                "frame": 0,
                "params": [
                    0,
                    15,
                    0,
                    0,
                    33.770896911621094,
                    -117.69420623779297,
                    20
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 10,
                "AltitudeMode": 0,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 6,
                "frame": 0,
                "params": [
                    0,
                    15,
                    0,
                    0,
                    33.770957946777344,
                    -117.69458770751953,
                    10
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 0,
                "AltitudeMode": 0,
                "autoContinue": true,
                "command": 21,
                "doJumpId": 7,
                "frame": 0,
                "params": [
                    15,
                    0,
                    0,
                    0,
                    33.771156311035156,
                    -117.69544982910156,
                    0
                ],
                "type": "SimpleItem"
            }
        ],
        "plannedHomePosition": [
            38.147918701171875,
            -76.42799377441406,
            200
        ],
        "vehicleType": 1,
        "version": 1
    },
    "rallyPoints": {
        "points": [
            [
                38.08412,
                -76.25459,
                20
            ]
        ],
        "version": 2
    },
    "version": 1
}

hongminy, 2018-11-19:
@hamishwillee I just uploaded the information

DonLakeFlyer, 2018-12-07:
Hmm, I've been debugging this. Currently concerned that given current architecture Auto-Load may be hopelessly broken. Still trying.

hongminy, 2018-12-07:
I have located that problem

DonLakeFlyer, 2018-12-08:
Ok, it's fixed. I was concerned about the crazy signalling that occurs in this scenario. But after going through all of it I believe it's ok. Also discovered that generic Rally Point loading had bugs as well.

DonLakeFlyer, 2018-12-08:
@hongminy Can you retest once the pull goes through and let us know how it goes?

PR #7076 (2018-12-08): AutoLoad: Not loading GeoFence or Rally points correctly
Fixes #7018
closed 2018-12-08 by DonLakeFlyer