mavlink/qgroundcontrol #7027: Can't flash Ardupilot for certain boards, file missing

mrpollo, 2018-11-20:
## Expected Behavior
When trying to flash my Pixhawk 4 with Ardupilot v3.6.1 QGC should be able to download a file from the Ardupilot firmware server and flash my hardware.

## Current Behavior
QGC Tries to download the wrong file to my Pixhawk 4, the file is also missing.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Boot QGC
2. Connect Pixhawk 4 via USB
3. Click on Settings (gears icon)
4. Click on Firmware (on sidebar)
5. Select ArduPilot Flight Stack from option pane on the right.
6. Select MultiRotor - ArduCopter v3.6.1 from dropdown

## System Information
When posting bug reports, include the following information
- Operating System: macOS Mojave 10.14.1
- QGC Version: 3.4.4
- QGC build: daily
- Flight Controller: Pixhawk 4
- Autopilot (with version): Ardupilot Copter 3.6.1

## Log Files and Screenshots

[image]
[image]
[image]
[image]

Discussed with @auturgy on [QGC discuss](http://discuss.px4.io/t/ardupilot-firmware-not-available-in-qgroundcontrol/8470)

DonLakeFlyer, 2018-11-20:
> QGC Tries to download the wrong file to my Pixhawk 4, the file is also missing.

The problem isn't quite the wrong file. That file would be correct if ArduPilot dropped a non-chibios version of firmware for a V5 board. But it does not, hence the file not found error. It only has chibios versions for V5 boards.

mrpollo, 2018-11-20:
Thanks everyone, lets keep this open until we resolve the issue (wherever that might be).

DonLakeFlyer, 2018-11-27:
I don't see a reason to spend the time to fix this. I've changed the dropdown such that ChibiOS builds are shown first a well as pulling the current ChibiOS build depending on port naming. NuttX builds are in the process of being phased out replaced by ChibiOS builds for everything. So going forward I think the current file not found error is good enough since in the future that should only happen if the ArduPilot release depot is screwed up.

closed 2018-11-27 by DonLakeFlyer