mavlink/qgroundcontrol #7032: Android: Daily build crashes at startup

fingadar, 2018-11-22:
The actual daily build on Android tablet crashes immediately after startup.
Tested with Android 6 and 7, same result.

The 3.4 release version works problem-free on same tablets.

Can I give any more useful information? I'm not an Android expert ;-)

DonLakeFlyer, 2018-11-22:
Hmm, something started going haywire around Nov 8th when crashes started to spike up. WIth 125 reports of java.lang.UnsatisfiedLinkError crash which means a native call is missing somewhere.

@dogmaphobic Anything ring a bell changing recently with android?

DonLakeFlyer, 2018-11-22:
And of course it doesn't repro for me.

DonLakeFlyer, 2018-11-23:
@fingadar Is this with a vehicle ready to connect to? Or just no vehicle there and boot.

fingadar, 2018-11-24:
That doesn't make any difference. I tested it with or without vehicle (UDP connected), with the same result. I think, qGroundControl crashes long before it tries to connect.

DonLakeFlyer, 2018-11-27:
@findagar Could you check if QGC has the USB port captured in application settings. If so, remove that settings. Also turn off all autoconnect in settings and try again.

DonLakeFlyer, 2018-11-27:
@fingadar Also what exact devices are you testing on and what OS version is on each one?

fingadar, 2018-11-29:
So, I did some short tests:

Medion X10302 with Android 6: Release version (3.4.4?) works, daily build crashes
Medion X10302 with Android 7: Release version works, daily build crashes
Stone aged Nexus7 (1st gen) with Android 4.4.4: both the release version and the daily build are working.

The X10302 is my CGS tablet that worked flawlessly with qGroundGontrol for 1 to 2 years now.

Crash means, the app is closed immediately after starting. It does not load the user interface, but at once the "app closed windows" ist shown. So I cannot do or change anything. 

In the installed release version, no USB port is configured. Deinstalling the Release version an reinstalling the Daily Build makes no difference.

What I noticed, the Daily Build does not ask for any permissions during installation. But even giving all permissions by hand makes no differences, it crashes anyway.

DonLakeFlyer, 2018-11-29:
@fingadar A new Android daily should have posted a few minutes ago. Can you give it a try?

DonLakeFlyer, 2018-11-30:
I believe this is fixed now. Hopefully @fingadar can verify. Otherwise Google Play Dashboard should tell us from crash reports.

fingadar, 2018-12-01:
I‘m very sorry, but same as before with the latest Daily Build. Crashes immediately after start.

DonLakeFlyer, 2018-12-01:
Can you check the build date/time. It's at the bottom of the General/Settings page.

DonLakeFlyer, 2018-12-02:
@fingadar I"m trying to track down the date of the last daily which did not crash. Can you try installing this: 
https://s3-us-west-2.amazonaws.com/qgroundcontrol/DonLakeFlyer/344443.apk

This is the daily from Oct 31 which I think is the last day it wasn't crashing.

And then install this, which I think is the first crashing build:
https://s3-us-west-2.amazonaws.com/qgroundcontrol/DonLakeFlyer/344519.apk

Let me know what happens with each.

fingadar, 2018-12-02:
Ok I tested it. You are right.

The 344443 version asks for permissions during installation and works. Connection via upd also works.

The 344519 version does not ask for any permissions during installation, and crashes immediately after starting.

And I'm very sorry, I can not tell you the build date/time of the last version of my last Playstore Daily Build version. It's crashing long before I can see the bottom of the General/Settings page :-). But Android's setting dialog tells me, it is 3.4.4-646.ga62fa768c, if that helps anyhow.

DonLakeFlyer, 2018-12-02:
> I can not tell you the build date/time of the last version of my last Playstore Daily Build version. It's crashing long before I can see the bottom of the General/Settings page :-)

Duh. Yeah!

DonLakeFlyer, 2018-12-02:
Can you try these two builds:
https://s3-us-west-2.amazonaws.com/qgroundcontrol/DonLakeFlyer/target-api-19.apk
https://s3-us-west-2.amazonaws.com/qgroundcontrol/DonLakeFlyer/target-api-26.apk

fingadar, 2018-12-03:
Both builds do not install. An error message with "incompatible application" comes up.

dogmaphobic, 2018-12-03:
You may need to manually uninstall the existing QGC. They have different signatures and Android won't allow one to override the other.

DonLakeFlyer, 2018-12-04:
@fingadar Don't bother testing those. We kind of know what is going on now. But still don't have a solution to fix.

fingadar, 2019-01-04:
@DonLakeFlyer Today, I installed the latest Daily Build, and it worked fine. No more crashes or starting problems yet. UDP connection worked. 
(I did not fly, but I do not think that there would be any problems)

DonLakeFlyer, 2019-01-04:
Thanks for reporting back. The crash was in video which was disabled and then finally fixed last week.

PR #7059 (2018-12-03): Android: Switch to sdk 25
Fixes #7032. Hoping this will fix the crash. I can repro crash now using pre-built daily build.
closed 2019-01-04 by DonLakeFlyer