mavlink/qgroundcontrol #7037: Add render type as initialization parameter

patrickelectric, 2018-11-26:
Some video hardwares does not support the same opengl engines that Qt tries to run, and this can result in an epic fail.
QGC should accept things like "--render=<render-mode>", where render-mode can be _desktop-opengl_, _opengl-es_, _software_ and etc.

jaxxzer, 2018-11-26:
I have only seen these epic fails on windows. There is a command line option like this that exists to do software rendering on windows. There are shortcuts installed to use those options (GPU SAFE MODE).

DonLakeFlyer, 2018-11-26:
Already supported. This is why QGC windows install adds three shortcuts for QGC. A regular one for opengl and the two others set the different non-OpenGL rendering options:

https://docs.qgroundcontrol.com/en/Support/Support.html

Look for Windows: UI Rendering/Video Driver Issues at the bottom

closed 2018-11-26 by DonLakeFlyer

patrickelectric, 2018-11-26:
And about linux and mac ?

DonLakeFlyer, 2018-11-26:
Never seen anyone reporting problems on those system. Linux tends to be an outdated driver problem.