mavlink/qgroundcontrol #7038: Joystick buttons on arducopter do not work

highfreq, 2018-11-26:
I am trying to calibrate my joystick in QGroundcontrol and all axis work correctly but buttons can't be assigned to any tasks. Label says "reserved by firmware".

I am using Arducopter 3.6 on a Cube 2.1.

regards,

Corrado

DonLakeFlyer, 2018-11-26:
@jaxxzer The original concept of ```FirmwarePlugin::manualControlReservedButtonCount(void)``` was because PX4 stack did something special with the ```MANUAL_CONTROL.buttons``` field. So QGC couldn't take over those for other things. Since PX4 no longer does that the PX4 plugin returns 0 which makes all buttons available to QGC. The ArduPilot firmware plugin still reserves all buttons though. Could you check the ArduPilot firmware side for use of ```MANUAL_CONTROL.buttons```. If it doesn't do anything with those could you change ArduPilot side as well to return 0 which will allow you to set button actions.

highfreq, 2018-11-26:
I do not have any skill to do what you ask. Maybe someone else could help?
Should i make a post on ardupilot discuss to see if any dev answer?

Corrado

DonLakeFlyer, 2018-11-26:
I'm not asking you to do anything. I am asking @jaxxzer to do something.

highfreq, 2018-11-26:
OOOpppsss sorry, didn't notice was @jaxxzer.......................sorry again

DonLakeFlyer, 2018-11-26:
No problem. Thanks for adding the issue.

jaxxzer, 2018-11-26:
AFAIK Sub is the only vehicle that makes use of these on the firmware side. (I can confirm if you want).

>  If it doesn't do anything with those could you change ArduPilot side as well to return 0 which will allow you to set button actions.

What do you mean allow to set button actions? Is there something implemented for handling the buttons on QGC side? I thought the autopilot was responsible for handling the button actions (ie this issue belongs in the firmware side if it is ever to be supported under the current MANUAL_CONTROL scheme).

jaxxzer, 2018-11-26:
For the record, I think it is insane that the autopilot is made responsible for handling button pushes on the joystick. I believe the GCS should be responsible for handling the button states, and send MAV_CMDs accordingly. I'd like to rework this some day, but it's not a small undertaking and will require changes to mavlink, autopilot firmware, and gcs code.

DonLakeFlyer, 2018-11-26:
QGC can assign actions to joystick buttons like arm/disarm, vtol transition and change flight mode. All handled internal to QGC. But in order to do that you have to tell it that the buttons are not reserved.

jaxxzer, 2018-11-26:
Ok, that's what I thought you were getting at, and that's the direction I'd like things to go, so great. I'll confirm on the apm side for copter etc.

highfreq, 2018-11-26:
you mean there is a chance this button thing is going to be fixed?

highfreq, 2018-11-26:
Do you thin the gimbal mount can be assigned too?

DonLakeFlyer, 2018-11-27:
> Do you thin the gimbal mount can be assigned too?

Not currently supported. Although I think you can move gimbal from the instrument panel camera ui page. Not sure. Haven't looked there in a while.

dogmaphobic, 2018-11-27:
No, gimbal control never made it to master. It's still sitting on a branch. It was last updated in May and then sort of died there.

highfreq, 2018-11-27:
So after buttons i guess we will have to open a request for gimbal support, without it is very difficult to use QGroundcontrol as a base software for custom ground stations

dogmaphobic, 2018-11-27:
With that said, gimbal control is on my TODO list. The main barrier has actually been UI rather than anything else.

jaxxzer, 2018-11-27:
It appears that ArduCopter makes no use of the buttons field in the MANUAL_CONTROL packet:
https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L1027

I'm confident this is the case, but @rmackay may confirm.

DonLakeFlyer, 2018-12-01:
Is it ok for me to allow buttons to work on ArduPIlot?

DonLakeFlyer, 2019-01-19:
Fixed now

closed 2019-01-19 by DonLakeFlyer