mavlink/qgroundcontrol #7045: Mag sensor inconsistent error with newer version of QGC.

DavidMackie, 2018-11-29:
Using the latest stable release of PX4 and latest stable release of QGroundControl, "Mag sensor inconsistent" errors pop-up as pre-flight checks fail.
When doing the calibration with a 2018 August QGC build, the problem disappears.

## Expected Behavior
After a successful Firmware update (PX4 stable), and a successful sensor calibration (for a Deltaquad airframe) the magnetometer inconsistency check has so far been working properly.

## Current Behavior
When using the latest QGC stable build, the inconsistency pops up and pre-flight checks fail, causing the arming to fail.

## Steps to Reproduce:
Steps to reproduce the behavior:

The orientation of the board must be 90 degrees roll, 90 degrees yaw.

1. With the latest stable version of QGC for Ubuntu flash the latest stable firmware.
2. Calibrate everything from fresh with the latest version of QGC for Android (Or Ubuntu)
3. Restart the drone
4. Mag sensor inconsistent and 90 degree heading offset is seen

## System Information
- Ubuntu 18.04
- QGC build which drops error: QGC Development HEAD:1bf12b5b3 2018-11-13
- QGC build which fixes error: QGC Development HEAD:9cb4f2ddc 2017-08-07
- Flight Controller: Dropix v2.1 from Drotek
- Primary Compass: Ublox NEO-M8N GPS + LIS3MDL compass
- Autopilot (with version): PX4 1.8.1 stable

## Additional related problem
Another issue that pops up is when the measured compass heading is about 90 off true heading. Calibration was done in relatively empty fields. The offset seemed to be consistent
This is either PX4 estimator related or calibration related.

## Log Files and Screenshots
Short test where the drone is pointing close to North direction:
[https://review.px4.io/plot_app?log=17da1d39-138e-4495-b304-4630c6c58a76](https://review.px4.io/plot_app?log=17da1d39-138e-4495-b304-4630c6c58a76)

A screenshot of the error when it pops up on on Android tablet:
[image]

Thanks for any help!

DonLakeFlyer, 2018-11-29:
The firmware is pretty much in full control of calibration. QGC just puts a pretty face on it. So this should go to firmware side to check first. There is a slight possibility that there is a QGC problem but I can't see how right now. My guess is that differences between old builds is just a fluke.

closed 2018-11-29 by DonLakeFlyer

DavidMackie, 2018-11-29:
I understand this seems weird. This is also the reason that it took us so long to find the error - checking the QGC version was one of the last things we checked.
With the latest version it works perfectly with the flight computer at an orientation only changing the yaw. As soon as we put the flight computer in a 90 degrees roll, 90 degrees yaw orientation it doesn't work.
Same firmware, same flight computer and same hardware. Repeated multiple times.

Thanks for the tip - I'll post this is PX4 firmware rep just in case.

DonLakeFlyer, 2018-11-29:
Could be a parameter meta data problem for the orientation parameter. Which would also be a firmware bug since the meta data comes from the firmware.

DonLakeFlyer, 2018-11-29:
> Could be a parameter meta data problem for the orientation parameter.

And that could be a difference between the two QGC versions.

DavidMackie, 2018-11-29:
@DonLakeFlyer we've now tried with 3 different firmware versions and done a diff check between the params file when it works and the params file when it has this issue there are no differences that are linked to this.

DonLakeFlyer, 2018-11-29:
I would say definitely not QGC side then.

DavidMackie, 2018-11-29:
Okay, thanks for your help