mavlink/qgroundcontrol #7049: UI broken when editing multiple non-default component params

edwinhayes, 2018-11-30:
## Expected Behaviour
When connected to a system comprised of multiple MAVLink components, the advanced parameter view should display a list of component IDs on the LHS (underneath the category groups for the default component).  Clicking on each component ID in turn should populate the table on the RHS with the parameters for that component.

## Current Behaviour
Clicking on the first non-default component shows the table of its parameters as expected.  Clicking on another non-default component does not change the table to show the parameters for the new component, the table appears not to update, so continues to show parameters for the first non-default component.

Clicking on one of the categories for the default component still has the expected behaviour, after which you can then click on another non-default component as expected.  So it's possible to view the parameters for many components by alternating between viewing parameters for the default component, and one of the non-default components.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1.  Be connected to a system with at least three MAVLink components (the default/autopilot, plus two others).
2. Navigate to the advanced parameter view.
3. Click on the menu item for one of the categories on the default component.  Works as expected.
4. Click on the menu item for non-default component A (say cID = 2).  Works as expected.
5. Click on the menu item for non-default component B  (say cID = 3).  Table fails to update.
6. Click on the menu item for one of the categories on the default component.  Works as expected.
7. Click on the menu item for non-default component B.  Works as expected.
8. Click on the menu item for non-default component A.  Table fails to update.

## System Information
When posting bug reports, include the following information
- Operating System: macOS
- QGC Version: 3.4.x
- QGC build: Daily from earlier this week.
- Flight Controller: Pixhawk2 + Other avionics 
- Autopilot (with version): ArduCopter 3.6.0

## Detailed Description

## Log Files and Screenshots
I will capture screenshots to explain this better when back in the office on Monday.

DonLakeFlyer, 2018-12-06:
Thanks for the detailed report

edwinhayes, 2018-12-11:
Now the UI updates as expected, but for some reason, some components parameters are displayed correctly, but for others, all the parameter values show up as zero.

See the two pics below; cID 240 and cID 38 are two instances of the same code running on a companion computer.  38 has two additional parameters defined (at runtime), but they are otherwise same, and its parameters are definitely not all zero.

[image]
[image]

The param messages are presumably getting to QGC, since it populates the list, but then something weird seems to happen.

edwinhayes, 2018-12-11:
I can capture some of the QGC log messages if you tell me which would be helpful; turning on all the ones related to "parameter manager" creates about a million pages of text which you probably don't want.

DonLakeFlyer, 2018-12-12:
Is it just the parameter editor displays that is wrong. Can you click on one to open the editor dialog? That will show the value for the parameter. If that is 0 as well, then there is some sort of protocol problem.

edwinhayes, 2018-12-12:
The affected values also appear as zero in the editor dialog.

DonLakeFlyer, 2018-12-12:
Can you capture a ParameterManagerLog and attach.

edwinhayes, 2018-12-12:
Attached, but a cursory search doesn't show anything related to any component other than cID 1?

[paramLog.txt](https://github.com/mavlink/qgroundcontrol/files/2670449/paramLog.txt)

edwinhayes, 2018-12-12:
Assuming this makes it in, I will test against the next daily build.

edwinhayes, 2018-12-13:
Here you go; with ParameterManagerVerbose1Log.

[paramLog.txt](https://github.com/mavlink/qgroundcontrol/files/2677676/paramLog.txt)

DonLakeFlyer, 2018-12-13:
If you look in the log you can see that for example the majority of the AIRSIDE_PORT values are coming across as 0. The log should you the raw values from the mavlink message. So not a QGC side problem.

edwinhayes, 2018-12-13:
The values are definitely not being sent as zeroes, because my software receives and decodes as expected; it must be some kind of protocol incompatibility between QGC and my software.  I suspect something to do with casting the value to the right type?  Hmm, but no idea why one of the instances works when the others don't since they should be the same binary.  I'll investigate on my end.

DonLakeFlyer, 2018-12-13:
Param values are stored in a struct based on type: mavlink_param_union_t

edwinhayes, 2018-12-17:
I've learned something new: Ardupilot has a different interpretation of the parameter protocol (as referenced in `APMFirmwarePlugin.cc` - line 247).  

On the system I'm testing against, while the flight controller is Ardupilot, all my companion computer modules should implement the parameter protocol per the original spec.

So I'm guessing the issue is that, for whatever reason, cID38 (which doesn't work) is being treated as an 'Ardupilot-like' device, while cID240 (which does work) is being treated as a 'PX4-like' device.

DonLakeFlyer, 2018-12-17:
That's shouldn't be the case. A FirmwarePlugin is associated with a Vehicle. All mavlink traffic should be routing through that single vehicle to the same FirmwarePlugin for futzing with on ArduPilot vehicles.

DonLakeFlyer, 2018-12-17:
> all my companion computer modules should implement the parameter protocol per the original spec.

You need to match the ArduPilot parameter spec in your companion code as well.

edwinhayes, 2018-12-17:
Is there some way I can efficiently obtain the raw messages which the parameter manager is decoding?  Because as far as I can tell, what I'm sending over the wire is identical except for cID, yet one component works and the other doesn't.

DonLakeFlyer, 2018-12-17:
Telemetry logs have the raw data in them.

edwinhayes, 2018-12-17:
@DonLakeFlyer - Yep, I was just hoping there might be some secret way of just getting the param messages in question, rather than trawling thought the telemetry log file!

So basically:

This message works:
`fd 19 00 00 c0 63 f0 16 00 00 02 00 00 00 08 00 07 00 4c 4f 47 5f 4c 45 56 45 4c 00 00 00 00 00 00 00 01 c1 f5`

And this one doesn't:
`fd 19 00 00 b3 63 25 16 00 00 02 00 00 00 0a 00 09 00 4c 4f 47 5f 4c 45 56 45 4c 00 00 00 00 00 00 00 01 8d f4`

Can anyone tell me why?  I feel like I'm going insane.

@hamishwillee - I'm talking to some of the AP devs about it currently.

auturgy, 2018-12-17:
I’ll ping @peterbarker to make it easier for him to find this.

hamishwillee, 2018-12-17:
Great. Just FMI, the component id in the first case is f0 (UDP bridge) and the second is Hex 25 - what sort of component is that?

edwinhayes, 2018-12-17:
It's another UDP bridge, just with a different comp ID assigned so you can tell the difference (there are a few such instances on our aircraft).  It's the same binary running in each case, the only difference is the parameter file it's started with.

edwinhayes, 2018-12-17:
So, upon further inspection, it turns out that `APMFirmwarePlugin.cc` has got a bunch of methods which _specifically_ handle cID240 differently: `adjustIncomingMavlinkMessage` for instance.  This is gonna be what that cID240 works and others don't.

edwinhayes, 2018-12-17:
@hamishwillee There is no need; these other cIDs fall within the gaps left in the MAV_COMPONENT enumeration for vendor specific component definitions, and they don't require any special treatment.

Ok, I understand the issue now: the APM firmware plugin 'adjusts' incoming parameter messages (except specifically for cID240) to fix the fact that AP isn't implementing the parameter protocol in the origin manner: `_handleIncomingParamValue`

The scope of the adjustment seems incorrect: it should only be adjusting messages from the flight controller (and any other components which is definitely known to have the same quirk) - it's not reasonable to go modifying messages from components you know nothing about.

DonLakeFlyer, 2018-12-17:
Seems like there should be something like a MAV_COMP_ID_USER1/2/... block reserved like are in other places.

> Ok, I understand the issue now: the APM firmware plugin 'adjusts' incoming parameter messages (except specifically for cID240) to fix the fact that AP isn't implementing the parameter protocol in the origin manner: `_handleIncomingParamValue`

Hmm, interesting. Never noticed that before. It's for the WiFi thingies. That said it kind of muddies the water as to what a ground station is supposed to do with respect to parameters from other components. Assume they implement the protocol as spec'ed or not? Or follow the protocol the vehicle is using.

DonLakeFlyer, 2018-12-17:
> The scope of the adjustment seems incorrect: it should only be adjusting messages from the flight controller (and any other components which is definitely known to have the same quirk) - it's not reasonable to go modifying messages from components you know nothing about.

I guess. But who knows what other ArduPilot bits expose themselves as another component and do params they same way main component does. Cameras? The rPi wifi APSync stuff? If not, then it could be spec'ed to say other components always follow spec. But I'm not sure if that is currently the case.

DonLakeFlyer, 2018-12-17:
@hamishwillee Can you discuss in the next mavlink dev meeting? If it works I would propose specifically saying only MAV_COMP_ID_AUTOPILOT1 has special handling and everything follows spec. Then it's an easy change in QGC to deal with that.

edwinhayes, 2018-12-17:
Re the MAV_COMP_ID_USER thing: I swear at some point in the past, common.xml actually had comments which denoted parts of the comp ID range as being reserved for vendor specific messages (I think maybe there was more like two ranges, each denoted as reserved for distinct purposes).  But that seems to be gone now?

If there are multiple devices out there with the other behaviour, seems like a pretty fundamental issue that needs to be addressed, since it breaks the concept of MAVLink compatibility pretty badly?

DonLakeFlyer, 2018-12-17:
> since it breaks the concept of MAVLink compatibility pretty badly?

Welcome to the world of QGroundControl...

hamishwillee, 2018-12-17:
> @hamishwillee There is no need; these other cIDs fall within the gaps left in the MAV_COMPONENT enumeration for vendor specific component definitions, and they don't require any special treatment.

Absolutely disagree. Please get an ID for components you plan to use because whatever gap you use may well be filled at some point. 

The component id is used for routing and filtering; so even if nothing is broken by this misuse, it is still likely that messages will be pushed around the network that don't need to be. 

I have not seen "MAV_COMP_ID_USER "

edwinhayes, 2018-12-17:
@hamishwillee It would be unreasonable for common.xml to include component definitions for things which have no meaning to most other users: "UDP loopback bridge to mysterious bit of proprietary software 12".  Per my comment to @DonLakeFlyer, I am sure there used to be ranges reserved for vendor definitions, what happened to those?

In my case, all my MAVLink components send heartbeat messages are required by the spec, so routing and filtering works just fine.

hamishwillee, 2018-12-17:
@edwinhayes Yes, but when component X uses that ID, they're going to get all your messages forwarded even if their at the end of a separate interface, even if they can't do anything about them. 

I don't know about user components. I have a sneaking feeling I've seen them though. Not sure. Have to think about whether they are a good idea. Probably OK as long as there are rules for how they are treated/not treated.

edwinhayes, 2018-12-17:
@hamishwillee True, but that's my problem: as an aircraft manufacturer, I can control the presence of other MAVLink devices.  If some new device appears in the common message set which uses an ID I'm already using, it's up to me to rearrange my allocations, or my products stop working properly.  However, the presence of defined ranges for vendor definitions does make this easier.

hamishwillee, 2018-12-17:
@DonLakeFlyer 

>  Can you discuss in the next mavlink dev meeting? 

That is planned for 3.5 weeks. We're skipping one because no one wants to have meetings on Xmas or Xmas eve. Can you wait?

I think I need more context in order to represent your "case"/what you expect.
I would expect any special handling for a component to be based on the particular component - irrespective of autopilot. If it needs to depend on a service/protocol, then it should be based on the specification.

If something isn't implementing the standard protocol(s) I would have though that would be based on the autopilot in use, as inferred from its heartbeat and via sysid to associated components. 

So I don't understand what special handling should be allowed on MAV_COMP_ID_AUTOPILOT1. 

And I'd be leaning towards @edwinhayes view that the trigger for managing behaviour for the parameter should not be component id for UDP bridge.

DonLakeFlyer, 2018-12-17:
> So I don't understand what special handling should be allowed on MAV_COMP_ID_AUTOPILOT1.

ArduPilot and PX4 send PARAM_VALUE mavlink messages in a different format.

hamishwillee, 2018-12-17:
@DonLakeFlyer Yes, but don't they both use MAV_COMP_ID_AUTOPILOT1?

Oh, I see, you're saying that you only want to allow modifications to the spec for messages that originate from an autopilot - not from a (say) Camera or UDP bridge?

Does that mean that a UDP bridge would need to do something special if passing info to QGC from ArduPilot (say)

DonLakeFlyer, 2018-12-17:
A ground station needs to know that and handle accordingly. Whereas what to do with component != MAV_COMP_ID_AUTOPILOT1 is unclear. 

Two options:
1) Follow the PARAM_VALUE format that is used for MAV_COMP_ID_AUTOPILOT1
2) Follow the PARAM_VALUE format which is in the mavlink spec

QGC Currently does 1 except for the UDP_BRIDGE component where it does 2. Which is a bit strange.

hamishwillee, 2018-12-17:
@edwinhayes 

> @hamishwillee True, but that's my problem: as an aircraft manufacturer, I can control the presence of other MAVLink devices. If some new device appears in the common message set which uses an ID I'm already using, it's up to me to rearrange my allocations, or my products stop working properly. 
However, the presence of defined ranges for vendor definitions does make this easier.

Well, to be a bit cheeky, you couldn't control the use of QGC on your network. 

But hopefully to be less annoying, if  you think user ranges will help (and I can see they might) then let's allocate a few items for this purpose and discuss in a PR in the mavlink/mavlink repo. How many IDs do you think you might reasonably need?

No guarantees they will get through, but that is the right channel for getting this change.

DonLakeFlyer, 2018-12-17:
> Yes, but don't they both use MAV_COMP_ID_AUTOPILOT1?

Yes. Firmware differences are identified from MAV_AUTOPILOT.

DonLakeFlyer, 2018-12-17:
The point of user ranges is to not conflict with known device types which may have special handling (like UDP_BRIDGE). The only possible on user range component is support for the parameter protocol.

DonLakeFlyer, 2018-12-17:
Yup

edwinhayes, 2018-12-17:
Really, I would like the APM vs PX4 param thing solved somehow; it erodes the whole attraction of 'MAVLink compatibility': will talk to @tridge and @peterbarker about it.

@hamishwillee Sure, I can probably do something tomorrow.

@DonLakeFlyer Thanks, that would be great.

hamishwillee, 2018-12-17:
OK, so this will be in the next Dev Meeting agenda [20190102](https://github.com/mavlink/mavlink/wiki/20190102-Dev-Meeting)

The most relevant bit is where I suggest the text:

> A GCS may provide support for different versions/variants of a protocol. 
>     A GCS is only expected to support versions for messages originating from an autopilot (component ID: `MAV_COMP_ID_AUTOPILOT1`). Messages from other components should be treated according to the most recent supported version of the specification. 

@edwinhayes 

> Really, I would like the APM vs PX4 param thing solved somehow; it erodes the whole attraction of 'MAVLink compatibility': will talk to @tridge and @peterbarker about it.

I could not agree more/please do. Along with Mission protocol it is the most damaging incompatibility between the two stacks for MAVLink. I know that ArduPilot need to evolve the parameter protocol for their needs. This should be done collaboratively with PX4 to represent the next (common) iteration of the protocol. I'm hoping @auturgy will be able to help make that happen.

> Sure, I can probably do something tomorrow.

Thanks very much.

DonLakeFlyer, 2018-12-17:
> A GCS may provide support for different versions/variants of a protocol.

I would not say that. Anything new should follow spec. The fact that ArduPilot does not is just history which I don't know the background of. I would just mention how ArduPilot works and the fact that anything moving forward should not work that way.

hamishwillee, 2018-12-18:
You misunderstand. I'm trying to say that this is how GCS should support spec evolution (spec deviations are a fact, but not something that can ever be "supported"). Even though this doesn't cover the deviation case per-se, it allows me to make it clear that a GCS only needs to support variation for MAV_COMP_ID_AUTOPILOT1.

Even so, this is challenging because:
- the version/component filtering feels a bit "out of scope" for the MAVLink definition. 
- A GCS might not implement the most recent version of the spec either, so this feels "false"

How about:

> Different autopilots may support a particular iteration of a microservice specification. A GCS that provides autopilot-specific support for a particular iteration will typically restrict this to only autopilot components (MAV_COMP_ID_AUTOPILOT1); for other components the GCS will use the most recent version of the specification it supports.

I'm also trying to capture what ArduPilot does different for params in https://github.com/mavlink/mavlink-devguide/pull/140 - what else should I say?

hamishwillee, 2018-12-18:
Perhaps this isn't MAVLink scope? It is something that should be send in the QGC dev guide.

edwinhayes, 2019-01-13:
Just confirmed that #7094 has fixed the issue which this issue originally described: I now have correct behaviour for both flight controller parameters (with their special ArduPilot treatment) and other component parameters (per the spec).  I will chase up the ongoing aspects (incompatibility between stacks etc) separately.

PR #7069 (2018-12-06): Parameter editor: Multiple components not displaying correctly
Fix #7049
closed 2019-01-13 by edwinhayes