mavlink/qgroundcontrol #7051: Vehicle defined configuration

sanderux, 2018-11-30:
As discussed on the last PX4 dev call i would like to propose Vehicle Defined Configuration.
The general concept is that vehicles through their airframe configuration (xml) can control certain options and standards.

To kick off the list i would propose:

- Cruise / Hover speed for missions
- ability to modify parameters or a list of exposed parameters in "normal user" mode 
- availability of flight modes / default flight mode
- ability to arm without gps fix
- default values to display in flight screen
- default / minimum takeoff altitude
- defaults for all options in the QGC settings screens with option to disable (e.g. virtual joysticks)
- minimum / maximum land altitude
- maximum mission distance
- max altitude AMSL
- turn / acceptance radius (perhaps to later plot an estimated flight path)
- availability of mavlink console
- availability of setting screens (such as power, tuning or airframe selection)
- availability of parameter reset option
- external mission validator link (should upload the plan file and show results)

These values could be stored in the airframes.xml file or an external location that should perhaps be queried now and then.

@DonLakeFlyer @dogmaphobic @dagar @LorenzMeier

dagar, 2018-11-30:
My first thought is that most of this should be well defined and controlled vehicle side at multiple levels.

Perhaps the airframes themselves should grow into something more declarative directly that's shared with QGC (rather than scripts and generated xml).

DonLakeFlyer, 2018-11-30:
> Perhaps the airframes themselves should grow into something more declarative directly that's shared with QGC (rather than scripts and generated xml).

Yeah, that's a really interesting way to do it.

OpenUAS, 2019-01-10:
Time to switch to Paparazzi maybe?

closed 2026-10-06 by github-actions[bot]