mavlink/qgroundcontrol #7065: periodic resets and QGC relinks to FC and lands back on the Setup summary page

dlarue, 2018-12-06:
Seriously?  Closing the issue(#7062) because I saw a jump from one port to the next?  It only does it on startup in the first 5 seconds and it stays reliably on the 2nd port.  When running QGC something is causing QGC to keep restarting the link.  Something is going on and it only happens when QGC is running.

dlarue, 2018-12-06:
It turns out the problem has nothing to do with that startup sequence and I figured the blue LED was the bootloader or something like that.  The firmware seems to work fine once up and running, even if on the next USB port assigned.  I've run ROS against it for almost an hour and it never blinked once and never caused an issue on the USB port.  I only started looking at the USB port when I saw the QGC logs showing ttyACM1.  For what ever reason, QGC has not had a problem in the last 30 minutes or so but I know that is intermittent since some times i've seen QGC do a number of restarts of the link in just a few minutes.  Yesterday, I could not get through the accelerator calibration for a number of iterations because it would restart the link and abort the calibration.  Yes it could be firmware but it feels like missing data validation checks.

DonLakeFlyer, 2018-12-06:
> Yesterday, I could not get through the accelerator calibration for a number of iterations because it would restart the link and abort the calibration.

All points to power problems across USB. Micro-usb is incredible prone to failure in my experience. Both the cords themselves and the connector on the device. Although I haven't seen a connector go bad on a Pixhawk device I've had many flaky ones on other things. And I've thrown out tons of micro-USB cables due to them being crap.

dlarue, 2018-12-06:
That could very well be and even though I made efforts to not touch anything I understand a fragile contact can still be an issue even when left still.  I will put on an external battery source and see if it becomes stable.  I had changed cables and ports but agree, the little connectors are poor and USB has no redundancy on any of the 4 lines...

dlarue, 2018-12-25:
using the latest PX4 Bootloader and Firmware as of 12/24/2018 I'm not seeing resets nor QGC disconnects.

closed 2018-12-25 by dlarue