mavlink/qgroundcontrol #7070: AirMap segfault on Linux

bartslinger, 2018-12-07:
My operating system is Ubuntu 18.04. I compiled QGC master @e1a290be2 including AirMap. AirMap is compiled using the instructions on https://github.com/airmap/platform-sdk, with qt-support enabled. I copied the libairmap-qt.so* files to `qgroundcontrol/libs/airmapd/linux/Qt.5.11.0`.

If I don't enter an API key, QGC does not crash. When I enter an API key, I get a segfault which seems to be triggered somewhere in `AirMapRulesetsManager::setROI`. If I disable this function, QGC does not crash. In that case I do get information like weather data for example.

This is what I see in the debugger:
[image]

dogmaphobic, 2018-12-07:
The Airmap SDK is incomplete (not finished). The only platform that currently works is macOS. Once they complete their SDK, I will add the remaining platforms. In the mean time, there is no Airmap for anything other than macOS.

closed 2018-12-07 by dogmaphobic

bartslinger, 2018-12-07:
@dogmaphobic I understand. Do you think the problem is in the SDK, or in QGC? If it's on the QGC side, I might be able to help out.

dogmaphobic, 2018-12-07:
QGC's support for Airmap is based on the previous SDK. The new one hasn't changed that much but it isn't finished. The original SDK had issues with Linux because it used some dynamic library that was in conflict with the one used by Qt (I can't remember the specific library). The fix for that would be for Airmap to link to it statically instead. The screen grab above shows something else. Probably some interface mismatch. Either way, until they finish it, there is little point in doing anything.