mavlink/qgroundcontrol #7092: *Feature Request* - ADS-B Overlay

micaheli, 2018-12-16:
Feature Title: ADS-B Overlay

Brief Description:
Airplane icons overlayed on the map as retrieved from an ADS-B receiver or IP address.  See http://www.flightaware.com or http://www.planefinder.net for an example of this.

Mission Planner has this - but it's poorly implemented and I tend to prefer QGC for basic missions due to the better mapping/interface.   This feature is the only one lacking that I would love to get.

This would require a property field for IP address/port at a minimum.   Additionally, supporting the different formats would be nice, but the main format on port 30003 is adequate - especially for an initial release/POC.  Many people set up ADS-B receivers on PiAware and connect their laptops directly to them for ADS-B on the field.   As I said, I'm already doing this with mission planner and it works OK.  The maps just aren't good enough for what I need and I'd like to see more information about the plane at a glance.  

Info under plane icon:
Tail number (if available)
Altitude / Speed
Distance from current location (if available)

micaheli, 2018-12-16:
> QGC already supports ADSB information coming in over mavlink from devices
> like this: https://uavionix.com installed on the vehicle.On Dec 16, 2018,

While that's super cool, that's not the intention for this feature
request.  This request is for the ground station to interact with a base
station ADS-B receiver.  It's a far less expensive option and helps prepare
for the 2020 ADS-B regulatory change for see/avoid.

It's a solution that works great for less than 100 bucks once, instead of
150 bucks per aircraft.  It also doesn't require your telemetry link to be
solid.

I really think this feature will help make QCG a single pane of glass for
all things aviation.

DonLakeFlyer, 2020-02-27:
QGC 4.0 supports this now

closed 2020-02-27 by DonLakeFlyer