mavlink/qgroundcontrol #7093: Goto point and Orbit circle stay on cancel

MaEtUgR, 2018-12-17:
Version: QGC Daily https://github.com/mavlink/qgroundcontrol/commit/5e63260bd4b40f66d433ac62b2f6d55e4de6d136

To reproduce:
1. Takeoff
1. As soon as the drone is in Hold mode press on the map
1. Choose to do a goto or an orbit
1. Cancel with the cross above the slider
1. **The planned goto point or movable circle stays present on the map until the next goto/orbit is invoked.**

[image]

Antiheavy, 2018-12-18:
old related issue: https://github.com/mavlink/qgroundcontrol/issues/6720

DonLakeFlyer, 2018-12-18:
Closing this one to keep #6720.

MaEtUgR, 2018-12-18:
👍

closed 2018-12-18 by MaEtUgR