mavlink/qgroundcontrol #7096: Automatic heading when moving to waypoint

jmcarvalhoo, 2018-12-19:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Please tell us what should happen as detailed as possible.

## Current Behavior
Please describe what happens instead of the expected behavior.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Create waypoints
2. Chose heading option
3. We can see that in the planner there is a heading but not on the options, we would like to get that heading to the mavlink plan so we could go to the next point facing forward.

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 18
- QGC Version: Last version as of 19 of December of 2018
- QGC build: stable
- Flight Controller: Bebob2

## Detailed Description
Provide further details about your issue/bug.
As we mention above we want automatic heading values like in the planner but in our file so the drone faces foward when moving to the next waypoint.

## Log Files and Screenshots
-  Two screenshots are in this link (https://imgur.com/a/z3xrQs7) one of the whole program and other of an example of plan.

DonLakeFlyer, 2018-12-22:
> 3\. We can see that in the planner there is a heading _but not on the options_

What does "_but not on the options_" mean?

jmcarvalhoo, 2018-12-26:
When I choose the option "Heading" on the plan it appears the direction that the drones is going to take but in the plan the Heading column is field with zeros as shown in this link: https://imgur.com/6Z5eKs7

We know that we can choose a manual heading but we want to be able to get  automatic heading values when choosing the next waypoint.

DonLakeFlyer, 2018-12-28:
How are you generating a text mission file as opposed to a json .plan file? I tested both Stable and latest Daily build and both correctly write the heading into the .plan file. 

Are you running the regular Bebop2 firmware or are you running something else on it. I'm not even sure that firmware supports setting a heading. That would be question for the firmware folks.

closed 2018-12-28 by DonLakeFlyer

DonLakeFlyer, 2018-12-29:
I will reopen if the information points to QGC problem. But so far QGC seems to be doing the right thing.

jmcarvalhoo, 2019-01-03:
Yes, it saves the .plan file but when we press the "upload" button it creates a .mavlink file as shown in the previous comment. (https://imgur.com/6Z5eKs7)

The issue still remains when choosing "Heading" option on waypoints it defines at zero even tho in the figure it automatically shows a Heading (https://imgur.com/a/uPzMdZt). We would like to know how to get the automatic heading so we can add to our plan.

DonLakeFlyer, 2019-01-03:
> Yes, it saves the .plan file but when we press the "upload" button it creates a .mavlink file as shown in the previous comment.

There isn't anything in QGC that would do that. So no idea where that is coming from. From the QGC side of things the heading is sent to the vehicle. Not much more I can do than that.