mavlink/qgroundcontrol #7100: *Feature Request* Import Obstacles

Geryones, 2018-12-24:
The import of a list of obstacles via KML would be really helpful. For the beginning, it would be enough, if the obstacles could be displayed even if they do not yet affect the flightpattern. 

Select KML File with all Obstacles 
[image]

Display the whole file content (here power lines in blue, anything else in red)
[image]

I'll try to implement it on my own, but every bit of help is more than welcome :)

jstrebel, 2018-12-24:
Hi, are you saying KML Import is not working at all?

Geryones, 2018-12-24:
No, KML import is working for a single Mission Item. At least I, was unable to import multiple items at once

DonLakeFlyer, 2019-01-02:
> No, KML import is working for a single Mission Item.

Which is a correct statement. You can't import KML which contains multiple polygons or polylines.

If you look at FlightDisplayViewMap you will see multiple examples of map overlays in there which are currently supported: AirMap, Orbit visuals and so forth. You need to add a new type from the which the data is provided by app.

Geryones, 2019-01-03:
Thank you very much. 
I found the type CustomMapItems in it, with the comment "Allow custom builds to add map items". Maybe I can start there? 
In the MissionController, I am able to create a QList<QGCMapPolygon> for all Polygons in a kml file, but I fail when I try to visualize them. I tried to add them to _visualItems, but that resulted in QGC crashing. 

How do I get from my List of polygons, to actually displaying them? They should not be interactiv or something, just static objects rendered on the map. 
What step am I missing?

DonLakeFlyer, 2019-01-03:
> How do I get from my List of polygons, to actually displaying them?

You write a bunch of code :) You're going to need to educate yourself on Qml. Look at how a GeoFence is displayed for examples.

Geryones, 2019-01-03:
This much I learned allready :) 
KK will do! Thank you

closed 2026-10-06 by github-actions[bot]