mavlink/qgroundcontrol #7107: landing pattern graphics not showing up in Fly screen in recent Daily build

Antiheavy, 2018-12-29:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Please tell us what should happen as detailed as possible.

## Current Behavior
Please describe what happens instead of the expected behavior.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1.
2.
3.
4.

## System Information
When posting bug reports, include the following information
- Operating System: [e.g. iOS 12.1, Windows 8, macOS Mojave 10.14.1, Ubuntu 14]
- QGC Version: [e.g. 3.4.4]
- QGC build: [e.g., daily, stable, self-built from source, etc...]
- Flight Controller: [e.g., Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.]
- Autopilot (with version): [e.g., PX4 1.8.1, Ardupilot Copter 3.6.1]

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

Antiheavy, 2018-12-29:
Hmmmm. This seems intermittent. I’ll try to find a reproducible order of operations.

DonLakeFlyer, 2018-12-31:
Code inspection didn't find anything. Could possibly be related to "Stop taking photos/videos" settings which are the newest changes.

Antiheavy, 2019-01-19:
i tried a couple times and couldn't figure out how to reproduce.

closed 2019-01-19 by Antiheavy

DonLakeFlyer, 2019-01-19:
Ok, bummer

Antiheavy, 2019-01-19:
thinking back on it I wonder if I was changing QGC and Firmware SITL versions and ran into an inconsistency.  When I tried to repo with normal QGC and normal Firmware I couldn't.