mavlink/qgroundcontrol #7116: Add warning/confirmation to autodelete video files.

jaxxzer, 2019-01-03:
I really don't believe that it is the duty of QGC to manage the space on a hard drive at all, that should be the user/os job.

But since QGC does manage this I recommend at least that the user is warned, and given the option to confirm instead of automatically and silently deleting what could be valuable data.

DonLakeFlyer, 2019-01-03:
Which kind of defeats the purpose of automatic delete. Because user will get asked over and over and over and over again about this.

jaxxzer, 2019-01-03:
Maybe the purpose of autodelete could evolve from silently obliterating data to friendly reminder/offering help to the user to manage their system?

helpdelete

DonLakeFlyer, 2019-01-03:
The reason for it was to not fill up mobile device storage which could happen real fast. It could be changed to warn if over certain limit. Still default to on for mobile only though.

jaxxzer, 2019-01-03:
I don't think I've ever encountered an application that removes user generated data without asking first.

Taking videos with your phone will fill the storage quickly. Even the camera app on your phone simply tells you 'storage is full'. And the user is expected to manage their system themselves, as they see fit.

Imagine if the iphone/android design choice was instead to silently and automatically delete your oldest pictures.

dogmaphobic, 2019-01-03:
I guess the main source of confusion is the terminology. Videos are meant to be stored on the camera. QGC "caches" the videos locally as a convenience. That is, when you start recording videos, QGC sends a command to the camera to do so and _in addition_, records and stores the stream locally as well. This is meant as a "cache" and not a final video (which is stored in the camera).

DJI does the same thing but uses a proper "cache" terminology:

[image]

DonLakeFlyer, 2019-01-03:
> I don't think I've ever encountered an application that removes user generated data without asking first.

My DVR does it very single day :)

> QGC sends a command to the camera to do so and _in addition_, records and stores the stream locally as well.

That's all going to depend on the camera you are using. DJI is truly a cache since they know what hardware they are running.

I think the concept was that you don't lose the current new saved video because you ran out of disk space along the way. What is more important an old video or the new one you are currently recording. Kind of hard to say. I'm fine with tossing the automatic thing if folks don't think it makes sense.

jaxxzer, 2019-01-03:
> I guess the main source of confusion is the terminology. Videos are meant to be stored on the camera. QGC "caches" the videos locally as a convenience.

Wow, that explains a lot in terms of the previous conversations on this.

This idea of a 'cache' was never our (my own, Blue Robotics) intention. We use QGC as the primary recording device. Our vehicle does not have the capacity to record videos locally. There is also the matter of transferring them to the computer. We also record directly on the computer as a convenience. We are using webcams and ip cameras without built in storage.

dogmaphobic, 2019-01-04:
I guess at the end of the day, it makes little difference. The main question becomes: what to do when you don't have space left? Mobile devices are slightly more graceful than desktop systems. When you run out of disk space on Windows, things tend to go downhill from there and your are locked up with a trashed system.

On the other hand, you must have something for the web camera to connect to, to collect the stream, encode it and stream it to QGC. You would have far better video quality saving it to local storage than what you get within QGC.

> That's all going to depend on the camera you are using. DJI is truly a cache since they know what hardware they are running.

Technically, QGC knows it too. If you have a camera, providing a full camera API implementation, it includes storage information. If all you have is a video stream, QGC can tell it by the lack of storage information. The Start/Stop recording video code path is different already for those two cases. We can certainly have different behaviors depending on the existence of "local storage". As usual, the UI would get more and more complex as a result.

DonLakeFlyer, 2019-01-04:
The original point was to try to prevent video save from ever failing. I'm perfectly fine with tossing the auto-delete stuff. For example telemetry saving is on by default (on the off case that you need one to debug a crash or fly away or something) and they just keep filling up storage until you go in there and delete them.

closed 2026-10-06 by github-actions[bot]