mavlink/qgroundcontrol #7125: Plan: Rework altitude mode terminolofy and ui (Writeup is WIP)

DonLakeFlyer, 2019-01-05:
**I'm still working on writing this up. Please no comments till I'm done and then I'll invite set of folks to discussion.**

The terminology around altitude mode: Rel, AMSL, AGL is still causing confusion. Specifically around whether Rel and AGL in airspace terminology can mean the same thing. That said getting this wrong can lead to crashes so taking another stab at this make a lot of sense.

I think there are other problems with the current ui as well as terminology. Specifically that is sticks what I consider power user options for setting altitude right in everyones face. All of which is confusing and clutters the ui. So even though a huge percentage of users will never use anything other than relative altitude they are exposed to the gory details of all the possibilities. And if they don't really know what they are doing and aren't careful/educated they can screw it up and crash into the ground.

From an internal functionality point of view there are four possible modes (using generic naming) for specifying an altitude/height value. The actual MAVLink frame used is in parantheses:
* **Relative** (MAV_FRAME_GLOBAL_RELATIVE_ALT) - Altitude specified is relative to the home altitude.
* **Absolute** (MAV_FRAME_GLOBAL) - Altitude specified is absolute height above mean sea level.
* **Terrain** (MAV_FRAME_GLOBAL) - This is a feature of QGC where you can specify a height above terrain and the actual value which is sent to the vehicle is the specified height plus the terrain height at that point as an Absolute AMSL value. It is specifically not the ArduPilot feature where the GCS can send terrain carpets to the vehicle for terrain following.
* **Terrain Frame** (MAV_FRAME_GLOBAL_TERRAIN_ALT) - This is a mode only supported by ArduPilot and not fully supported by QGC. It will only show up if you load a mission which is already using this MAV_FRAME in a WAYPOINT item. You cannot create mission in QGC which used this frame. QGC does not support sending terrain carpets to the vehicle. If you don't know what this is then don't worry about it. You will never see it.

So the question becomes how to expose these to the user in a way that is usable and less prone to screw ups.

**Current ui:**
[image]
[image]
[image]

**Proposed ui tries to solve the following problems:**
* Don't stick power user altitude modes in all users faces. It does this by placing them behind a hamburger menu you can click on to get to them.
* More specific terminology on modes
* Toss the altitude field slider thing which still isn't quite ready for prime time

By default you get relative altitudes. For most users this is as far as they will go. It's clean and simpler.
[image]

If you click the hamburger you can pick another mode. When you change mode both the field label above the numeric value and the small help text below the value change.
[image]

What you get for Absolute:
[image]

What you get for Terrain:
[image]

What you get for Terrain Frame:
[image]

**So the points of discussion are:**
* Is the new ui layout better: 
  * Altitude mode label + hamburger instead of radio buttons
  * Altitude value
  * Altitude mode help small text
* The text used for each altitude mode label and help text as well as the text in the popup menu.

DonLakeFlyer, 2019-01-06:
@hamishwillee @AndKe @antiheavy @sundancemediagroup @jacobsenmd @auturgy @dogmaphobic  

FYI: Looking for comments.

I personally like a lot better than what was there before. It's simpler for most users. I'm not super happy about the double hamburger menus but I can live with it.

auturgy, 2019-01-06:
Nice!
The terrain reference frame text isn’t super clear, but “Height above SRTM” would be worse, and if you can’t set it in a mission it’s not really critical.

hamishwillee, 2019-01-07:
1. Presenting Relative to Home "first and foremost" (hiding the complexity) is better.
1. Nested hamburger is indeed ugly
   - as you say, we can live with it.
   - but are there other options? swipe whole section sideways or arrow toggle between options?
1. The top altitude text is slightly inconsistent: Altitude, Above Mean Sea Level, Above Terrain, Terrain Frame - I would have though the first option woudl be "Above Home"
1. I don't think sub-text should exist in cases where it just repeats top level.
1. Why quote AMSL value in subtext for "Terrain" but not "Relative"?
1. Terrain QGC vs onboard frame unclear. 

How about following text/sub-text options (I have assumed "Altitude" is implied):
   * Above Home | Actual AMSL altitude []
   * Above Mean Sea Level | - 
   * Above Terrain (QGC data)| Actual AMSL altitude []
   * Above Terrain (onboard Map) | Actual AMSL altitude []

Antiheavy, 2019-01-07:
> * Presenting Relative to Home "first and foremost" (hiding the complexity) is better.
> 
> * Nested hamburger is indeed ugly
>   
>   * as you say, we can live with it.
>   * but are there other options? swipe whole section sideways or arrow toggle between options?
> 
> * The top altitude text is slightly inconsistent: Altitude, Above Mean Sea Level, Above Terrain, Terrain Frame - I would have though the first option woudl be "Above Home"
> 
> * I don't think sub-text should exist in cases where it just repeats top level.
> 
> * Why quote AMSL value in subtext for "Terrain" but not "Relative"?
> 
> * Terrain QGC vs onboard frame unclear.

I had similar thoughts to all these statements.

Antiheavy, 2019-01-07:
Apologies if this is too fundamental a question or out of scope of this issue: 
Do you want QGC to support mixed altitude frames within a single mission?  In my opinion there shouldn't really be a need for mixed altitude frames within a mission.  This may allow a simpler interface for selecting altitude frame globally for the mission and then just add little reminder text in each Waypoint Description which altitude frame you are in for the mission.  Altitude frame for the whole mission could even be switched dynamically by the user.

hamishwillee, 2019-01-07:
@Antiheavy 

^^^^ That is an excellent point.

DonLakeFlyer, 2019-01-07:
> The terrain reference frame text isn’t super clear, but “Height above SRTM” would be worse, and if you can’t set it in a mission it’s not really critical.

I agree. I haven't been paying too much attention to that one yet since it's only sort of supported. I'll take another swing at it.

> 3\. I would have though the first option woudl be "Above Home"

My feeling is that a large portion of the user base has no concept of anything other than "Altitude". Since they flat of mostly flat terrain this just works for them without even known that this is really a relative altitude above the home position altitude. Hence I just call it "Altitude" too keep things simple for the majority of users.

> * Above Home | Actual AMSL altitude []

Two things: 1)The actual AMSL altitude for Relative altitudes isn't really clear. In order to truly know that you need to know the home position which isn't necessarily know until you actually connect and arm a real vehicle. 2) Even if it could be done reliablity I think this will just add confusion to the non power user who has no idea what AMSL is.

> Do you want QGC to support mixed altitude frames within a single mission?

The mavlink spec supports it so QGC has to support it. It could be possible for custom builds though. It is also possible to set the Altitude Mode in Mission Start which would then be used throughout unless you change it. As it currently stands the altitude mode from the previous waypoint is used for newly added waypoints.

> Most pilots cannot fathom the difference between MSL and AGL. This is where the KISSS principle is needed. Keep it short, simple, specific.
> MSL is not simple, nor is it ever specific to any UAS pilot's operation.

QGC needs to support the MAVLink spec which supports the MAV_FRAME_GLOBAL as described above. This is essentially an AMSL altitude as described by spec. Custom builds can turn this ability off, but normal QGC needs it.

DonLakeFlyer, 2019-01-07:
> The mavlink spec supports it so QGC has to support it. It could be possible for custom builds though. It is also possible to set the Altitude Mode in Mission Start which would then be used [...]

There is a possibility of maybe hiding this capability behind a setting. But I think that would require a larger audience for discussion. I have been bitten many times with QGC trying to simplify things by hiding or not supporting and then some comes along with a really important reason as to why that was the completetly wrong decision.

AndKe, 2019-01-07:
I think both the current and the new design looks good, except: 
this https://user-images.githubusercontent.com/5876851/50728645-173c6c80-10e2-11e9-8b71-51d8d5b98516.png
should also mention "relative" instead of just "Altitude" , to be like the others.

And yes, there are good reasons to have do missions with mixed altitude types, QGC is quite powerful as a planning tool, but different altitude frames are important to us who fly big planned surveys, on high places or deep valleys, never knowing exactly where a suitable parking/landing spot is, but arriving with a well made main mission.

in short: do not reduce the professional features of planning, rather keep it, and work on improving the flight monitoring to a more professional level.

DonLakeFlyer, 2019-01-07:
@AndKe Do you ever use AMSL though?

hamishwillee, 2019-01-07:
Another option along the lines of @Antiheavy proposal is to put the altitude selection options that are displayed and the default ordering somewhere generic. The default setting could be Altitude (above home) and people like @AndKe could choose whatever other options they liked.

AndKe, 2019-01-07:
I think both the current and the new design looks good, except: 
the new should never type out the setting with small text below only. 
------------------------
50     m 
above terrain
-----------------------

seems just dangerous, I would rather see
------------------------
50     m AMSL
above mean sea level
-----------------------
- never leaving any doubt.

so yes, like  @hamishwillee  says, the devfault could very well be
(hamburger)----------
50     m REL
relative to home
-----------------------

AndKe, 2019-01-07:
Yes, Perfect !  impossible to mistake.

DonLakeFlyer, 2019-01-07:
Ok, let me try reworking that.

DonLakeFlyer, 2019-01-07:
Latest pull:
[image]
[image]
[image]

hamishwillee, 2019-01-08:
Me too, though 
1. I still quite like the idea that the selectable options could be chosen in settings: https://github.com/mavlink/qgroundcontrol/issues/7125#issuecomment-452093833
2. What about for Terrain Frame? (vs Above Terrain)
3. For the last one, why do we quote actual AMSL altidude? Is it just because that is what we are really sending and we want people to know?

DonLakeFlyer, 2019-01-08:
> 1. I still quite like the idea that the selectable options could be chosen in settings:

The problem is with options is that whatever you choose as the default is what is thought to be the only feature set available. My take if that the other Altitude mode options are hidden away enough that people won't futz with them by mistake. Hence no need to hide them even further.

> 2\. What about for Terrain Frame

I'm not too worried about that on since it's a bit of a more out of the ordinary usage pattern. You would have to create a mission in MP and then load in QGC.

> 3\. Is it just because that is what we are really sending and we want people to know?

Correct.

edwinhayes, 2019-01-15:
> I'm not too worried about that one since it's a bit of a more out of the ordinary usage pattern.

The `MAV_FRAME_GLOBAL_TERRAIN_ALT` behaviour is core functionality for our customers and is one of the main reasons that we are forced to use Mission Planner instead of QGC today.  I hope I will be able to assign an engineer to implement support for this in QGC sometime this year.

The new scheme seems good; except I'd like to see "above terrain" changed to something to more clearly discriminate its behaviour (flying above where QGC thought the terrain was at planning time) from "terrain frame" (realtime estimation of terrain clearance using whatever sensors the flight controller has at its disposal).

DonLakeFlyer, 2019-01-16:
> The `MAV_FRAME_GLOBAL_TERRAIN_ALT` behaviour is core functionality for our customers and is one of the main reasons that we are forced to use Mission Planner instead of QGC today.

Is that with a rangefinder for terrain or do you need terrain height maps sent to the vehicle while flying?

edwinhayes, 2019-01-16:
@DonLakeFlyer - Both; for instance we have one customer who uses a laser RF, plus feeding of both regular SRTM and custom DEM data to the vehicle in flight.

DonLakeFlyer, 2019-01-17:
Closing this out since I think it about as good as it's going to get at least for a while.

closed 2019-01-17 by DonLakeFlyer

fingadar, 2019-05-09:
The use of a rangefinder with qGroundControl and Ardupilot would be perfect. In my case also, this is  is one of the main reasons that we are forced to use Mission Planner instead of QGC today.

DonLakeFlyer, 2019-05-10:
> The use of a rangefinder with qGroundControl and Ardupilot would be perfect. In my case also, this is is one of the main reasons that we are forced to use Mission Planner instead of QGC today.

Not sure what you are referring to. That case should work fine since it doesn't require reponding to TERRAIN_REQUEST messages. At least as far as I understand.

fingadar, 2019-05-13:
I did some tests yesterday. The copter was equipped with a Lightware SF11 and Arducopter 3.6.8. The starting point was at 400m ASL. I defined a simple mission with one waypoint app. 16m lower, at 384m ASL.
Flight 1: Defined the waypoint in qGroundControl as 40m above terrain, takeoff height 40m: The Copter took of to 40m AGL, then proceeded to the waypoint and descended down to 26m AGL. Mission Planner showed the simple mission as waypoint with 426m ASL absolute.

DonLakeFlyer, 2019-05-13:
That sounds like you are using "Above Terrain" for the altitude. That means it will be calculated from the terrain database and just an AMSL alttiude to that. If you want to use Lidar with ArduPilot you need to use "Terrain Frame" instead.

fingadar, 2019-05-13:
I do not have a "Terrain Frame" option (Android Daily or OSX Daily)?
Just "Relative To Home", "Above Mean Sea Level" and "Above Terrain"

DonLakeFlyer, 2019-05-14:
Ah, crap. That will only show up if you are loading a mission which is already using it. I did that because QGC does not yet support TERRAIN_REQUEST and vehicles having ability to sense terrain height are few so it was less confusion that way. Otherwise folks might think QGC supports TERRAIN_REQUEST which it doesn't yet.

You can work around it with Show All Values though (from the hamburger menu on the right) and set the frame directly to MAV_FRAME_GLOBAL_TERRAIN_ALT. Pretty nasty from a UI perspective. But if you are desperate to use QGC for this it should work.

I have it on my list to support TERRAIN_REQUEST as coming soon. Once that is done, then Terrain Frame will show up in builds all the time.

fingadar, 2019-05-15:
The work around is no problem, thank you. But in surveys I do not have this possibility, I think?

DonLakeFlyer, 2019-05-15:
> But in surveys I do not have this possibility, I think?

Correct unfortunately. Let me think if I can make some sort of workaround.

rizvanovic, 2021-10-04:
This never got implemented? It is very tedious to have to edit waypoints manually to change to different altitude types. This seems like a solid addition to QGC.