mavlink/qgroundcontrol #7133: RTK: NTRIP Corrections

Naterater, 2019-01-07:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----
## Expected Behavior
I want to be able to access RTK corrections from my NTRIP provider. Mission Planner allows me to do this, however I see nothing about NTRIP in QGC.

## Current Behavior
Unavailable

## Steps to Reproduce:
N/A

## Detailed Description
I believe Mission Planner's code utilizing NTRIP is here: https://github.com/ArduPilot/MissionPlanner/blob/b26324bdc7e9abd87eb7e9c02bfc11f7184a9a51/ExtLibs/Comms/CommsNTRIP.cs

dogmaphobic, 2020-02-14:
Is the data provided by an NTRIP server the same you get from a regular RTK base? (sorry, I briefly looked up NTRIP and got lost rather quickly with the cacophony of mud like explanations).

highfreq, 2020-04-25:
Would it be possible to add ntrip corrections from an internet server (we have subscroption) and inject it trough mavlink? Basically same as MP

maxhofidrone, 2020-07-24:
Hi,
are there any news of this feature? I would be very interested for using data of a NTRIP server **and** QGC?

bys1123, 2020-10-15:
I saw few Chinese survey drone company need this feature too.

AlexKlimaj, 2023-10-03:
I would also like to see this implemented

ChinaMan28, 2023-10-03:
Implementing this would literally change the UAS Mapping/Survey industry. I to support this.

KnightHawk06, 2024-08-09:
Any movement on this ticket? Would be very useful to be able to connect to remote RTK via NTRIP in QGC.  https://www.u-blox.com/en/product/u-center currently supports this.

KnightHawk06, 2024-08-15:
@AlexKlimaj Thanks for pointing me to the client. I was able to get it setup but I'm having issues connecting.

[image]

mavlink-routerd -e 127.0.0.1:14550 -e 127.0.0.1:14551 -e 127.0.0.1:14552 /dev/ttyUSB0:57600

I've started mavlink-router forwarding the 3 ports that I'm connected to, I am able to connect to 14550 and 14551 but not 14552.

 ./build/polaris-client-mavlink 
Waiting for MAVSDK connection: 127.0.0.1:14552

This is with QGC v4.3 on Ubuntu 22.04

Any ideas? I have been able to connect to my Polaris account in ucenter with my login credentials, there's not anything else needed to connect to the api key is there?

AlexKlimaj, 2024-08-16:
Where is mavlink router running and where is polaris running? Also 57600 baud is likely too low.

KnightHawk06, 2024-08-16:
Mavlink router, Polaris and QGC are all running on the same Ubuntu box pointing to the local host (127.0.0.1)

I'll try a higher data rate.

KnightHawk06, 2024-08-16:
I tried updating the RFD 900x-US radio settings to connect at 115200 and 460800 but I'm not able to get QGC to connect via UDP to mavlink-router at those rates for some reason, it will only connect at 57600.

Are there any logs that I can look at? I'm only seeing this message below from the polaris client. 

Waiting for MAVSDK connection: 127.0.0.1:14552

It looks like I'm connected to the port in QGC and I see some message content using netstat.

KnightHawk06, 2024-08-16:
I was able to get QGC to connect at 115200, you have to update SER_TEL1_BAUD to 115200 if you update the baud rate on the radios. (assuming you're connected to TEL1).

AlexKlimaj, 2024-08-16:
So QGC should connect to port 14550 and polaris should connect to 14552.

KnightHawk06, 2024-08-16:
Correct, QGC is connecting on 14550 but Polaris just prints the "waiting for MAVSDK connection" message