mavlink/qgroundcontrol #7140: No heading hold option in survey mode

raviss2k18, 2019-01-11:
Hello,
I made a survey grid in qgroundcontrol but could not find heading hold option. I have used Mission planner also which provides such option.
Does qgroundcontrol provides no such features. Basically i want keep heading of drone fixed during survey.

thank you

tops4u, 2019-05-12:
Same Issue here... Thanks!

DonLakeFlyer, 2019-05-14:
Not currently supported

tops4u, 2019-05-15:
@DonLakeFlyer This would really be a very nice to have. How can I open a Feature Request? I assume it would need an additional Field in the Survey where a YAW Angle could be spcified (alternatively it could even be a checkbox that adaptes the YAW Angle to the Grid Angle).

DonLakeFlyer, 2019-05-15:
No need, this issue can track it. The problem with adding new options to Survey as it currently stands is that more options will cause it to collapse under it's own weight. It's already way too complicated to use. It needs a full uyi rework before new stuff can be added.

copterspace, 2020-07-07:
I think, currently simple workaround could be to download .plan file, generated by Survey in QGC to local computer, and write simple python/pearl script, which modifies “params”[4] in all mission points to desired angle instead of “null”. Then upload this modified mission file.
If anyone did that already - please share your experience. If not - I’ll try to do this myself and share mine…
[image]

DonLakeFlyer, 2020-07-07:
If you edit a saved Survey plan file directly you are putting the survey out of sync with the mission items. I'm maybe 90% certain this is ok.

The safest way is to Instead round trip the Survey it to the vehicle and back (keep your original survey plan as well). This will turn it into a pile of waypoints. Save that and then you are correct that NAV_WAYPOINT.param4 is yaw.

copterspace, 2020-12-09:
I posted my solution here: https://github.com/copterspace/QGC_plan_fix
Tested on jmavsim.

closed 2026-10-06 by github-actions[bot]