mavlink/qgroundcontrol #7144: Report: Daily build (14.01) segmentation fault on start

MaEtUgR, 2019-01-14:
## Expected Behavior
QGC starts. With the build from 06.01.2019 it all works fine.

## Current Behavior
I get:
```
maetugr@ubuntuvm:~$ ./QGroundControl.AppImage 
Settings location "/home/maetugr/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "*Log.debug=false\n"
GStreamer debug output: "/home/maetugr/Documents/QGroundControl/Logs/gst/gstreamer-log.txt"
Error loading text-to-speech plug-in "speechd"
Failed to set Geoclue positioning requirements. Geoclue error: org.qtproject.QtDBus.Error.InvalidObjectPath
QGCPositionManager error 2
MAVLinkLogManagerLog: MAVLink logs directory: "/home/maetugr/Documents/QGroundControl/Logs"
Map Cache in: "/home/maetugr/.cache/QGCMapCache300" / "qgcMapCache.db"
Segmentation fault (core dumped)
```

## Steps to Reproduce:
1. Download the latest daily build AppImage from: https://docs.qgroundcontrol.com/en/releases/daily_builds.html now (14.01.2019)
`wget https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/master/QGroundControl.AppImage`
2. `chmod +x QGroundControl.AppImage`
3. `./QGroundControl.AppImage`

## System Information
When posting bug reports, include the following information
- Operating System: [Ubuntu 18.04 hosted on VMWare Windows 10 x64]
- QGC build: [Daily from 14.01.2019 09:00 european time]

DonLakeFlyer, 2019-01-16:
@dogmaphobic Did I see you just fix a crash with qgcApp reference? Is that at boot?

dogmaphobic, 2019-01-16:
> Is that at boot?

No, it was at shut down. Something was accessing `qgcApp()` after it was nullified.

DonLakeFlyer, 2019-01-16:
Hmm, all working fine for me.

MaEtUgR, 2019-01-17:
Strange. I just tried to download again and it all works fine. I still had the file I last downloaded and that now also worked. I have no idea why it didn't work last time even after a lot of retries but the old QGC version worked. Could it have something to do with the state if you run it on a fresh computer? I can test that but close the issue for the time being. Sorry for any hassle, just wanted to help.

MaEtUgR, 2019-01-17:
I just tested with a fresh Ubuntu 18.04 setup and that's how it crashes:
First I get this error because the modem stuff is Ubuntu stock:
[image]
Here you see the complete history. After having my user in the correct group and removing modemmanager, reboot it still seg faults:
[image]

MaEtUgR, 2019-01-17:
And to verify it is not my VM setup I tested the "old" daily from 06-01-2019 starts and after I used that version once the new one also starts:
[image]
[image]

DonLakeFlyer, 2019-01-17:
@MaEtUgR Awesome, thanks for digging into this. I'll get a new VM going and repro.

DonLakeFlyer, 2019-01-17:
I can get it to crash now.

DonLakeFlyer, 2019-01-17:
Thanks for the help

closed 2019-01-17 by DonLakeFlyer

MaEtUgR, 2019-01-18:
Nice! I verified it's fixed with daily 17-01-2019. Thanks 👍