mavlink/qgroundcontrol #7147: Problem install on Android

yeechongyeung, 2019-01-17:
## System Information
When posting bug reports, include the following information
- Operating System: [ Windows 10, Qt5.11.0   for Android  ]
- QGC Version: [lastest]
- QGC build: [ self-built from source]

## Detailed Description
after install the compile result apk, only show the log window, and whilte block.  

## Log Files and Screenshots
[image]

W QGroundControl: (null):0 ((null)): QQmlApplicationEngine failed to load component
W QGroundControl: (null):0 ((null)): qrc:/qml/MainWindowNative.qml:12 module "QtQuick.Window" is not installed
W QGroundControl: qrc:/qml/MainWindowNative.qml:13 module "QtQuick.Dialogs" is not installed
W QGroundControl: qrc:/qml/MainWindowNative.qml:12 module "QtQuick.Window" is not installed
W QGroundControl: qrc:/qml/MainWindowNative.qml:13 module "QtQuick.Dialogs" is not installed

DonLakeFlyer, 2019-01-17:
This could be just about anything. First step is I would try to build/install android examples from Qt using your setup.

closed 2019-01-17 by DonLakeFlyer

DonLakeFlyer, 2019-01-18:
My other comment would be that I haven't heard anyone get android builds to work on Windows yet. QGC devs for android is mostly done in OSX. And the CI build system uses Linux to build.