mavlink/qgroundcontrol #7155: 3.4.4 stable version crashes on Mac OS X 10.10.5

wittyapps, 2019-01-18:
Unable to launch version 3.4.4 on Mac OS X 10.10.5, I've included the error reported to Mac.

Process:               QGroundControl [1041]
Path:                  /Applications/QGroundControl.app/Contents/MacOS/QGroundControl
Identifier:            org.qgroundcontrol.qgroundcontrol
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           QGroundControl [1041]
User ID:               501

Date/Time:             2019-01-18 10:32:33.721 -0800
OS Version:            Mac OS X 10.10.5 (14F2511)
Report Version:        11
Anonymous UUID:        CD372122-98D8-2F28-F00E-C71AAA85B09C

Sleep/Wake UUID:       764BCEC9-D811-4EC7-AAF4-D9E3D5FC932D

Time Awake Since Boot: 1900 seconds
Time Since Wake:       1700 seconds

Crashed Thread:        0

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000002, 0x0000000000000000

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Symbol not found: __os_log_default
  Referenced from: /Applications/QGroundControl.app/Contents/MacOS/../Frameworks/QtCore.framework/Versions/5/QtCore (which was built for Mac OS X 10.11)
  Expected in: /usr/lib/libSystem.B.dylib
 in /Applications/QGroundControl.app/Contents/MacOS/../Frameworks/QtCore.framework/Versions/5/QtCore

DonLakeFlyer, 2019-01-18:
I believe we moved the SDK used for QGC to require El Capitan or greater. Your OS is just too old for QGC to run.

closed 2019-01-18 by DonLakeFlyer

pklapperich, 2019-03-12:
This is a documentation bug, then. [Current documentation](https://docs.qgroundcontrol.com/en/getting_started/download_and_install.html) says the OS X 10.8+ is supported. 

> **Mac OS X**

>Install QGroundControl for Mac OS X 10.8 or later:

> 1. Download QGroundControl.dmg.
> 2. Double-click the .dmg file to mount it, then drag the QGroundControl application to your Application folder.

DonLakeFlyer, 2019-03-22:
@hamishwillee Can you update docs?

hamishwillee, 2019-03-25:
@pklapperich Thanks - update requirement to macOS 10.10 in https://github.com/mavlink/qgc-user-guide/commit/e53e1aa412562feab7028325cb19c521883ab306