mavlink/qgroundcontrol #7175: Android QGC does ot connect to Dragonlink TX USB port (transparent Mavlink telem)

Hwurzburg, 2019-01-21:
Don, as requested the Win QGC connect to DL tX USB port console log...
also, here is the USB info Android reports for a Chibios board that does connect vs the info for DragonLink TX (which is a composite device)...there are only a couple of differing values

Device Info for Chibios board: 
Device Path: /dev/bus/usb/001/003
Device Class: Communication Device (0x2)
Vendor ID:  0483
Vendor Name (reported):  ArduPilot
Vendor Name (from DB):  SGS Thomson Microelectronics
Product ID:  5740
Product Name (reported):  F35Lightning
Product Name (from DB):  not found

Additional Info 
Interface #0 
Class: Communication Device (0x2)
Endpoint: #0
Address        : 0x82 (10000010)
Number         : 2
Direction      : Inbound (0x80)
Type           : Intrrupt (0x3)
Poll Interval  : 255
Max Packet Size: 8
Attributes     : 000000011
Interface #1 
Class: Communication Device Class (CDC) (0xa)
Endpoint: #0
Address        : 0x01 (00000001)
Number         : 1
Direction      : Outbound (0x0)
Type           : Bulk (0x2)
Poll Interval  : 0
Max Packet Size: 64
Attributes     : 000000010
Endpoint: #1
Address        : 0x81 (10000001)
Number         : 1
Direction      : Inbound (0x80)
Type           : Bulk (0x2)
Poll Interval  : 0
Max Packet Size: 64
Attributes     : 000000010

and for DL TX:

Device Path: /dev/bus/usb/001/002
Device Class: Miscellaneous (0xef)
Vendor ID:  1fc9
Vendor Name (reported):  NXP
Vendor Name (from DB):  NXP Semiconductors
Product ID:  00083
Product Name (reported):  VCOM Port
Product Name (from DB):  not found

Additional Info 
Interface #0 
Class: Communication Device (0x2)
Endpoint: #0
Address        : 0x82 (10000010)
Number         : 2
Direction      : Inbound (0x80)
Type           : Intrrupt (0x3)
Poll Interval  : 2
Max Packet Size: 16
Attributes     : 000000011
Interface #1 
Class: Communication Device Class (CDC) (0xa)
Endpoint: #0
Address        : 0x01 (00000001)
Number         : 1
Direction      : Outbound (0x0)
Type           : Bulk (0x2)
Poll Interval  : 0
Max Packet Size: 64
Attributes     : 000000010
Endpoint: #1
Address        : 0x81 (10000001)
Number         : 1
Direction      : Inbound (0x80)
Type           : Bulk (0x2)
Poll Interval  : 0
Max Packet Size: 64
Attributes     : 000000010

DonLakeFlyer, 2019-12-24:
@Hwurzburg Is this still a problem with latest dailies?

DonLakeFlyer, 2020-02-08:
@Hwurzburg ?

Hwurzburg, 2020-02-09:
I havent changed my DL firmware in a while so am not sure its even an issue any more...but will set up next week and test and also ask someone who upgrades regularly to confirm...