mavlink/qgroundcontrol #7176: UVC video not working for some devices.

Rickyg32, 2019-01-21:
## Expected Behavior
There are a number of users running the FUAV UVC device on an android device. The device is not seen by Qgroundcontrol yet it is seen by other applications running on the Android device. It's been described to me as an issue related to how Native Android apps versus C++ written apps see the device. However with UVC devices becoming commonplace, the ability to watch UVC video from these devices would be beneficial.
https://www.aliexpress.com/item/SKYDROID-UVC-Dual-Antenna-Control-Receiver-OTG-5-8G-150CH-Full-Channel-FPV-Receiver-W-Audio/32959321778.html?spm=2114.search0604.3.23.7fbb1b19ZGqNuu&ws_ab_test=searchweb0_0,searchweb201602_2_10065_10068_10130_10547_319_317_10548_10696_453_10084_454_10083_433_10618_431_10139_10307_537_536_10059_10884_10887_100031_321_322_10103,searchweb201603_56,ppcSwitch_0&algo_expid=07294b54-c095-4fa7-bdef-f0288b384575-3&algo_pvid=07294b54-c095-4fa7-bdef-f0288b384575&transAbTest=ae803_5

## Current Behavior
The UVC device is not seen as an available option in Qgroundcontrol. It is seen by other apps like Tower for instance. It is also seen by Qgroundcontrol on Windows 10 devices. But not on the Android version of Qgroundcontrol.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Plug in the FUAV UVC device.
2.Open Qgroundcontrol
3.Go to the section in the config for Video Source and see if it's there.
4.

## System Information
When posting bug reports, include the following information
Android version 8.0.0

## Detailed Description
No other information can be provided as the device doesn't show so there are no log references or screen shots that can be provided.

## Log Files and Screenshots
-

dogmaphobic, 2019-01-21:
UVC support in QGC is done through Qt. Qt for Android does not support UVC. For this to work, you would either need to "fix" Qt, or write a specialized UVC handler for Android.

Rickyg32, 2019-01-21:
I realize that but Don asked me to open the record.

dogmaphobic, 2019-01-21:
Yup, I needed this for a different project using an HDMI UVC dongle. I ended up resorting to a Microsoft Surface tablet because Android did not support it.

HyGy, 2019-04-12:
I have an eachine rotg01 pro uvc receiver. It does not work with qgroundstation. First id does not work with tower, but then I found a recompiled version and now it is working with tower.

here is the version that supports it with tower
https://discuss.ardupilot.org/t/tower-is-crashing-when-using-uvc-stream-android-6-0/15621/6

Maybe it helps to make it work with qgroundcontrol too.

dogmaphobic, 2019-04-12:
UVC is not supported under Android. This is a Qt issue and won't be solved until they do.

HyGy, 2019-04-12:
> UVC is not supported under Android. This is a Qt issue and won't be solved until they do.

If it is not supported how is it working with tower?

dogmaphobic, 2019-04-12:
Tower is a real Android app, likely written by real Android programmers, which none of us are. QGC is a Qt application. It's a cross platform tool set that allows us to have one single source base for all targeted platforms.

taileron, 2019-10-06:
there is a great workaround available the android app usb camera now with rtsp server support and QGC offers a very good low latency rtsp client -> works perfect now !!!

IngoTheMan, 2019-10-13:
Do you guys use a OTG usb hub to get both telemetry and video to qgc? Isnt it either telemetry or video?

taileron, 2019-10-15:
... I mostly use ESP 8266 with better antennas and on the ground a usb powered dualband router 2.4G to ESP and 5G to QGC so the rtsp stream comes by 5G for lowest latency

Rickyg32, 2019-10-15:
> there is a great workaround available the android app usb camera now with rtsp server support and QGC offers a very good low latency rtsp client -> works perfect now !!!

@taileron is there a specific USB app your referring to.

IngoTheMan, 2019-10-15:
> ... I mostly use ESP 8266 with better antennas and on the ground a usb powered dualband router 2.4G to ESP and 5G to QGC so the rtsp stream comes by 5G for lowest latency

And the router is powered from a powerbank or something?

taileron, 2019-10-16:
with router (powerbank) it´s a few hunded meters (connection was always solid)
Groundstation consists of QGC connected to Router by 5Ghz and a Quanum Half Overlord FPV receiver connected to a VS8100 rtsp video server.
The wifi in the plane causes a few rc frame losses (interaction between 2.4 GHz channel hopping and fix channel) but rc always remains solid. For 5GHz Wifi and FPV - no interference at all - if the Quanum is used.

If ROTG0X connected to USB Camera (Android APP) Esp8266 is in AccessPoint Mode (Tablet is connected to the wifi of the esp directly) only 120 .... 150m range similar to ROTG0X

Rickyg32, 2019-10-16:
So if I read the above and understand it correctly. You have a standard a standard video transmitter in the craft. IT transmits to the Overlord receiver
The receiver is outputting video to the VS8100 rtsp server.
The VS8100 rtsp server is routing the RTSP to the Wifi router.
The Wife router is routing to the...this is where I get lost. Is the ESP8266 being used as a wifi access point for the VS8100 rtsp server to route its signal to the tablet. Then the tablet takes the RTSP signal and uses it in QGC.. If so whats the ROTG0X being used for...puzzled.

Rickyg32, 2019-10-16:
I think the ESP8266 needs to be combined with an arduino doesn't it. I have never used it so not sure.

taileron, 2019-10-16:
can be connected instead of 433 / 915 MHz telemetry.
Offers 921600 b/s for extremely smooth hud with:
mavlink stop-all
mavlink start -d /dev/ttyS1 -b 921600 -r 20000 -m onboard
in sd/etc/extras.txt

e.g. Pixracer offers a dedicated slot preconfigured to 921600 b/s

Rickyg32, 2019-10-16:
doh now I get it. Your using the ESP8266 on a pixracer for Telemetry routing...the light jut came on.

CamGenius, 2022-03-08:
I have a workaround that I thought I would post here. The "nExt Camera" app works with the UVC camera then you can press the button for PiP mode and then you can then go ahead and use QGroundControl with the video stream on top. Also some tablets like my Tab S6 allow apps to be side by side and prefer this because then I have a large video view next to QGroundControl which seems to scale nicely to being smaller. This setup works better for me over RTSP because then I don't get the latency.

taileron, 2022-03-08:
In Android I last used ip camera pro app to convert from uvc to rtsp that also worked but with a lot of clicking. In the meantime I got MacOs86 installed on my tablet and can either plug in the ROTG directly or use my long range version with the external rtsp encoder VS8100. I can no longer tolerate the very poor picture quality of the ROTG receiver via USB and prefer to accept a little latency ... However, the nExt Camera App seems to offer a high level of convenience when used on a smartphone where the image quality is not so noticeable.

closed 2026-10-06 by github-actions[bot]