mavlink/qgroundcontrol #7180: Parachute command needed

moreba1, 2019-01-22:
Parachute command
We need using parachute in an emergency such as Stall, and we can’t use the designed plan . also we can not using remote controller beacuse range of remote controller is low . for this reason we need a botton for parachute in fly (The speed of operation is needed at specific times).
I before this button in c-astral c3p software .

vascodx, 2019-01-31:
I suggest adding a deploy parachute command to the mission planning section. I am using arduplane and can't use qgroundcontrol because of this missing command. 
Thanks in advance.

DonLakeFlyer, 2019-01-31:
@vascodx Parachute command is available in missions:
* Add a waypoint
* Change the command for the item either by clicking on the "Waypoint" title or selected "Change Command" from the hamburger menu on the right.
* Change to the "Safety" category
* The parachute command will be in there.

moreba1, 2019-02-01:
@vascodx

selsect ardupilot and fixedwing in mision start menu

AlexandrShipovsky, 2022-07-13:
Is there a parachute release button at any given time? It would be convenient for those who fly without RC control