mavlink/qgroundcontrol #7182: Don't automatically enable joystick after calibration.

jaxxzer, 2019-01-22:
I think 'Click Next to enable joystick' should be changed to 'Joystick calibration complete, check enable joystick to begin sending joystick updates to the vehicle.'

Calibrating and enabling output should be two independent steps that the user triggers.

jaxxzer, 2019-01-22:
This also provides a little bit of self-documentation on what the enable joystick box does.

DonLakeFlyer, 2019-02-04:
Don't understand how this is better or why current mechanism is a problem.

MaEtUgR, 2019-07-24:
Maybe a solution would be to have a "Save calibration" and a "Save Calibration and enable joystick" button at the end of the procuedure? btw there are to some extent related changes: #7595

DonLakeFlyer, 2019-07-24:
I think for 90% of users the fact that it enables is goodness. I wouldn't add additional ui button complexity for the 10% case. If there is confusion to the fact that the joystick enables itself. Then when you click the last Next just pop up a message that says "Joystick enabled". Then folks who don't want that can just disable.

closed 2026-10-06 by github-actions[bot]