mavlink/qgroundcontrol #7185: QGCS Crash on Startup in Debug Mode - QT Creator Tool

patilakash, 2019-01-23:
The QGCS is crashed when I run in Debug Mode and works good in Release Mode. The error is attached
[image]

dogmaphobic, 2019-01-23:
Instead of a screen grab, can you collect the stack trace so we can have some idea where it all started?

patilakash, 2019-01-23:
Hi,

Application window output is

23:18:54: Starting
D:\Working\build-qgroundcontrol-Desktop_Qt_5_11_0_MSVC2015_32bit-Debug\debug\QGroundControl.exe...

QML debugging is enabled. Only use this in a safe environment.

ASSERT: "data.ptr->ref.isStatic()" in file kernel\qmetaobject.cpp, line 152

Regards

Akash Patil

DonLakeFlyer, 2019-02-04:
Without a stack trace no much we can do here

DonLakeFlyer, 2019-02-04:
Reopening since I just this myself. It's windows only.

closed 2019-02-04 by DonLakeFlyer