mavlink/qgroundcontrol #7197: PARAM tuning via RC lots of issues

tuloski, 2019-01-30:
The overall param tuning via RC on qground is a nightmare.
I don't know if it is still supported but it is not working properly at all.
I don't know if the problem is QGC or the autopilot.

The problems I see:
- Even if I remove the advanced setting "Set RC to param", the parameter remains binded to the RC channel.
- It is not clear at all what the scale is.
- Min/max seems not to be used, or used in a weird way.

For example I had a parameter with center value 10, min 5 and max 20, scale 1. When the RC was at lower position (1000us) the param became 9. When at higher position (2000us) the param became 11. I thougth it was a scale issue, so I changed to 10 having weirder results.

To be rethinked and revamped completely IMO.

DonLakeFlyer, 2019-02-04:
> To be rethinked and revamped completely IMO.

Yeah, this UI is kind of ancient history compared to the reset of QGC. It should still work though. This feature isn't controlled by QGC though. It just sends the values to the firmware and the vehicle side is completely responsible for implementing it. So if anything isn't working with respect to the values you sent that would be a firmware bug.

> * Even if I remove the advanced setting "Set RC to param", the parameter remains binded to the RC channel.

In order to clear RC to Param settings you need to select Clear RC to Param from the tools menu.

tuloski, 2019-02-04:
> So if anything isn't working with respect to the values you sent that would be a firmware bug.

To understand if it's a QGC problem of PX4 problem I need to understand the protocol that is used between QGC and copter for this kind of setting. Any link/info? Is it following some MAVlink standard or is it an ad hoc implementation in PX4?

closed 2026-10-05 by github-actions[bot]