mavlink/qgroundcontrol #7204: mission transfer error when uploading to arducopter 3.6.5 (and 3.6.4)

maciek01, 2019-02-02:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Please tell us what should happen as detailed as possible.

should be able to upload this mission

## Current Behavior
Please describe what happens instead of the expected behavior.

upload fails with "mission transfer error. retry transfer. Error: Vehicle returned error: X/Param5 has invalid value (MAV_MISSION_INVALID_PARAM5_X). Param5: nan."

when u reload mision from vehcle it shows just a WP 3.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. upload mission file to QGC
2. select upload button
3. error pops up
4.

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 18.4
- QGC Version: HEAD:f585fd801 2019-01-23 23:41:10 +0000
- QGC build: daily
- Flight Controller: Pixhawk 2
- Autopilot (with version): Ardupilot Copter 3.6.5

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

[wilmington3.plan.txt](https://github.com/mavlink/qgroundcontrol/files/2824829/wilmington3.plan.txt)
[image]

DonLakeFlyer, 2019-02-03:
Hmm, there is a MAV_CMD_SET_CAMERA_MODE command in there somehow. That is not supported by ArduPilot. Could you have initially created this plan while not connected to a vehicle and had the Vehicle Info set to PX4 Pro at the time and then checked the Mode checkbox in the Camera section.

Note that the Camera section of waypoints will not show up if the Vehicle is set to ArduPilot. It will only show up if the vehicle is set to PX4.

DonLakeFlyer, 2019-02-03:
This does show a problem with the Camera section of the Waypiont editor though. Even if the vehicle type is ArduPilot is the plan has camera changes in it then the Camera section should become visible. Otherwise there is no way to turn this off.

DonLakeFlyer, 2019-02-03:
After pull #7205 goes into a daily build you'll be able to remove that setting from the Plan.

maciek01, 2019-02-03:
@DonLakeFlyer this mission was built initially with arducopter connected w/o any cam control commands. firmware type was always set to arducopter.
Recently i edited the mission w/o copter attached and added the cam angle dispositions as i am testing out my new tarot gimbal. this what resulted in the error.

BTW since i could not proceed i rebuilt the mission from scratch adding the cam angles as i was adding new WPs. This new mission uploaded fine. So my guess is that editing existing WPs and adding gimbal angles is what exposed the bug.

DonLakeFlyer, 2019-02-04:
I went through the code and don't see any way for it to programmitcally change the set camera mode setting. Only code to do that is initiated from the user clicking in the ui. So there is a possibility of a bug here, but code inspection doesn't show it.

closed 2019-02-04 by DonLakeFlyer