mavlink/qgroundcontrol #7211: RTSP video stream option is not showing in build but shows in installed QGC

Rhonins, 2019-02-04:
Hi,
When I am running the QGC I have installed with setup, RTSP video stream is popping in video stream options. I can see my video stream.
[https://imgur.com/a/A5DILvt](url)

But when I run my own build Taisync option pops-up and RTSP video stream is not showing in drop down menu. Also I am not able to see rtsp video stream under TaiSync.
[https://imgur.com/a/lAzrNgd](url)
[https://imgur.com/a/pr1KoRY](url)
Is that a bug?
or
How can I figure that out?

dogmaphobic, 2019-02-13:
None of the images above are working. Just place them directly to the issue instead of a link.

I'm not sure I fully understand the problems you are having. Keep in mind that when you enable Taisync, video streaming is handled automatically. That's why the video stream settings are not available.

roseyangyu, 2019-02-26:
I have same issues

roseyangyu, 2019-02-26:
[image]
this is QGC.AppImage download from https://docs.qgroundcontrol.com/en/getting_started/download_and_install.html. RTSP video stream is popping in video stream options. But I have build that with same version.It only has video stream disabled and easycamera.
[image]
Please help me to kown this is bug or errors during me buildding

dogmaphobic, 2019-02-26:
You don't have GStreamer installed.

closed 2019-02-26 by dogmaphobic

roseyangyu, 2019-02-27:
thanks for your advice.I should install GStreamer [follow here](https://github.com/mavlink/qgroundcontrol/tree/master/src/VideoStreaming)

NejatRmz, 2019-05-17:
I have followed the instructions [here](https://github.com/mavlink/qgroundcontrol/tree/master/src/VideoStreaming). When i run the code it works fine for windows but in the android build i get only "No Video Available" option. you can see capture [here](https://imgur.com/Ji3IUjs)

Sathiyanarayanan18, 2023-07-12:
Iam Also facing Same Issue, Can you please help to solve this issue

bismahA, 2023-08-02:
Same issue
Did someone find any solution?

orca741, 2024-03-14:
Same issue!
Please advice