mavlink/qgroundcontrol #7213: QGC fails to connecrt to Intel Aero Drone

RohanKurane, 2019-02-05:
When opening the QGC app on workstation while connected to the Aero hotspot, I see the following
```think@penguin:~/Downloads$ ./QGroundControl.AppImage 
Settings location "/home/think/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "*Log.debug=false\n"
QGeoPositionInfoSourceGeoclueMaster(0x2b02750)
Failed to set Geoclue positioning requirements. Geoclue error: org.qtproject.QtDBus.Error.InvalidObjectPath
QGCPositionManager error 2
MAVLinkLogManagerLog: MAVLink logs directory: "/home/think/Documents/QGroundControl/Logs"
Map Cache in: "/home/think/.cache/QGCMapCache300" / "qgcMapCache.db"
```

I am running on Ubuntu 18.04 with openssl version 1.1.1
```OpenSSL 1.1.1  11 Sep 2018```

I do not have these issues on Ubuntu 16.04 with openssl version 1.0.2g

@lucasdemarchi 
@LorenzMeier

RohanKurane, 2019-02-05:
I installed 1.0.2q but still cannot connect QGC to the Intel-Aero. 
Is there anything else I need to do ?

DonLakeFlyer, 2019-02-05:
None of those messages have anything to do with not connecting. Not sure what it takes to connect to AeroDrone other than I hear it being somewhat futzy and complicated.

lucasdemarchi, 2019-02-05:
@RohanKurane What the openssl version have to do with not being able to connect? 

@DonLakeFlyer  it's actually very simple: there's a UART inside and mavlink-router turns this into UDP/TCP - i.e. there's only 1 more component to look after.

RohanKurane, 2019-02-05:
@lucasdemarchi 
I noticed in another issue that someone had SSL error messages while launching QGC. 

All I do is connect my workstation to the AERO-xxx hostspot. Then launch QGC on workstation. Shouldnt this automatically connect to the vehicle from QGC ?

patrickelectric, 2019-02-05:
@RohanKurane QGC is deployed with openssl1.0, probably ubuntu 18.04 is using a different version. Try to install version 1.0 and update LD path before calling QGC AppImage in the terminal.
```
export LD_LIBRARY_PATH=/usr/lib/openssl-1.0
```
The path for openssl-1.0 can be different.

RohanKurane, 2019-02-06:
I can try this again. 
@patrickelectric  fwiw...I did downgrade to openssl 1.0.2q but same error. Maybe my PATH was not updated

patrickelectric, 2019-02-06:
@RohanKurane, You need to update the LD_LIBRARY_PATH as I said here:
https://github.com/mavlink/qgroundcontrol/issues/7213#issuecomment-460750484
And after that, you should execute QGC AppImage without problems in the same terminal.

patrickelectric, 2019-02-06:
@DonLakeFlyer 	I just tested the last AppImage here and QGC can download the firmware and flash it without problems.
There is some ssl errors though. 
```
qt.network.ssl: QSslSocket: cannot resolve SSLv3_client_method
qt.network.ssl: QSslSocket: cannot resolve SSLv3_server_method
```

DonLakeFlyer, 2019-02-06:
> You need to update the LD_LIBRARY_PATH as I said here:
> [#7213 (comment)](https://github.com/mavlink/qgroundcontrol/issues/7213#issuecomment-460750484)

Like I said that should not be needed using latest builds.

closed 2019-06-25 by RohanKurane