mavlink/qgroundcontrol #7217: Add option to select language

dogmaphobic, 2019-02-10:
QGC currently sets the language based on what the OS reports. There is no way for the user to select a different language. A user might have the OS set to their native language but would prefer QGC in English (or some other combination).

QGC should ask the user what language to use (on first boot) and also provide an option to change within its settings.

dogmaphobic, 2019-03-31:
This is now in master.

closed 2019-03-31 by dogmaphobic

evexoio, 2019-04-10:
@dogmaphobic I just installed qgroundcontrol v3.5.0 and I got turkish as a default language,
I'm in turkey but I don't have turkish as system language, and none of my keyboard layouts are turkish as well.

I can't find an option to change the language to english.

Where can I find the language option?

dogmaphobic, 2019-04-10:
It's on the daily builds: https://github.com/mavlink/qgroundcontrol/pull/7300

evexoio, 2019-04-10:
@dogmaphobic I just went to the settings and searched for it as in the screenshot on the PR, couldn't find the language setting...

[image]

dogmaphobic, 2019-04-10:
>It's on the daily builds

You are running the _stable_ build.

zhangxh2000, 2019-07-15:
> This is now in master.

I used the latest Master，but when I change to another language,nothing happened, and I found this log "QGCApplication.cc:437-"Error loading Qt localication for "zh_CN“",I builded the QGC myself，Does I missed something when building?

datwwe, 2020-08-12:
> > This is now in master.
> 
> I used the latest Master，but when I change to another language,nothing happened, and I found this log "QGCApplication.cc:437-"Error loading Qt localication for "zh_CN“",I builded the QGC myself，Does I missed something when building?

Is your problem solved yet? I'm having just the same problem when build QGC by Qt5.6.12. Thanks for your reply!